arendst/Tasmota #1408: Easier configuration of multi-channel switch / PWM devices

emontnemery, 2017-12-21:
To make configuration simpler and also enable auto-discovery by home-assistant or by a potential Domoticz plugin (#1344) would it make sense to:
- Allow "template" configuration of some LED-dimmer devices (act as a switch, act as a number of individual dimmers, act as an RGB dimmer, etc)
- Listen to separate topics for each switch / PWM-channel
- Publish the configuration in a way that can be understood by home-assistant: https://home-assistant.io/docs/mqtt/discovery/

Big4SMK, 2018-01-06:
I have taken a first shot at making (multi channel) switches announce themselves to home assistant in https://github.com/arendst/Sonoff-Tasmota/pull/1534

I don't have LED dimmer devices, so I have no option to work/test those.

emontnemery, 2018-01-19:
That's really nice!

Does home-assistant specify a way to trigger retransmit of the discovery messages?

emontnemery, 2018-01-21:
@Big4SMK Do you plan to add support also for discovery of sensors and buttons?
I think button should be announced as "binary_sensor" if SwitchTopic is set to 1 or 2?

Big4SMK, 2018-01-22:
I was indeed removing the retained messages that way. Note that @arendst completely rewrote the code and slightly changed the functionality. From what I see in https://github.com/arendst/Sonoff-Tasmota/commit/1538d30c96a542d6126a6827ac94d45411058ffe he added support for lights, but not (yet) for sensors, and I don't really feel like adding code that I cannot test. Maybe you could change the code, verify, and create a merge request?

emontnemery, 2018-04-02:
Closing this, home-assistant discovery is working fine.

closed 2018-04-02 by emontnemery