arendst/Tasmota #1411: Sudden loss of WiFi connection on 3 different devices

BernRos, 2017-12-22:
I have 3 devices, Sonoff POW, Sonoff TH10 and a Sonoff Switch 4Ch Pro. I use all 3 devices with http-requests via openHAB. After flashing with TASMOTA, all 3 devices worked fine within my wireless LAN. But suddenly (maybe after a router reset) all 3 devices lost the WiFi connection at the same time. Now I'm not able to bring them online again. I tried to power off and on again, I tried to reconnect with the built in Access Point. Nothing works, they will not (re-)connect to the network.
Do I really have to flash them again or am I doing something wrong? And if I have to flash the devices again, how can I prevent this failure to happen again?

BernRos, 2017-12-22:
Still not sure about the reason, but after a night of switching the power off, all devices are working again. Sorry for bothering you with this stupid thing...
Anyhow, one question is left for me: Does the sonoff device reconnect automatically after a Router shutdown/reset?

arendst, 2017-12-22:
Check your wificonfig options. Once all is configured I prefer setting it to option 5 (Wifi wait for network availability)