arendst/Tasmota #1415: Sonoff Pow - change of topic for power information?

nquick, 2017-12-22:
I had my devices on 5.6 and was running well and able to get all the power information off the devices into pimatic using the tele/sonoff_dev#/ENERGY. 

I've just updated to 5.10 (OTA this time) and noticed my energy stats not updating any more? When I look at the information page I see this:

tele/sonoff_con02/SENSOR = {"Time":"2017-12-22T19:41:20","ENERGY":{"Total":13.691,"Yesterday":0.257,"Today":0.184,"Period":0,"Power":55,"Factor":0.98,"Voltage":227,"Current":0.247}}

Pimatic can't see this new information now and so I just get a UNKNOWN in the value. I don't have 5.6 to go back to, to check the settings against it.

Did it change?

Frogmore42, 2017-12-23:
From the release notes:
/* 5.10.0b

 * Add support for PZEM004T energy sensor to be enabled with define USE_PZEM004T in user_config.h

 * **Change Sonoff Pow Energy MQTT data message** and consolidate Status 8 into Status 10

The data used to be sent to ENERGY, looks like it now goes to SENSOR with ENERGY contained within.  Look at this section in the Pimatic docs for help in how to get it to understand:
Accepts JSON message with hierarchy from: [https://pimatic.org/plugins/pimatic-mqtt/](url)

pascalsaul, 2017-12-25:
Use this in Pimatic:
name = ENERGY.Voltage
topic = tele/pow/SENSOR

closed 2018-05-08 by stale[bot]