arendst/Tasmota #1418: MQTT topic differ when posting update after manual switching and MQTT command switching

alakdae, 2017-12-22:
When switching manually the topic is correct as set in FullTopic:
00:43:16 APP: Button 1 immediate
00:43:16 MQT: v1/devices/sonoff_4ch_1/stat/RESULT = {"POWER1":"OFF"}
00:43:16 MQT: v1/devices/sonoff_4ch_1/stat/POWER1 = OFF
00:43:16 APP: Button 2 immediate
00:43:16 MQT: v1/devices/sonoff_4ch_1/stat/RESULT = {"POWER2":"ON"}
00:43:17 MQT: v1/devices/sonoff_4ch_1/stat/POWER2 = ON

But when switching through MQTT command the reply is like this:
00:44:20 RSL: Received Topic v1/devices/sonoff_4ch_1/cmnd/Power3, Data Size 2, Data ON
00:44:20 RSL: Group 0, Index 3, Command POWER, Data ON
00:44:20 MQT: stat/sonoff_4ch_1/RESULT = {"POWER3":"ON"}
00:44:20 MQT: stat/sonoff_4ch_1/POWER3 = ON

Where can I set the reply topic for MQTT commands to be the same as FullTopic and same as publishes from manual switching?

arendst, 2017-12-23:
#1317

alakdae, 2017-12-23:
Thanks for the solution.
Maybe maintainer should consider adding a note in user_config.h that MQTT_CLIENT_ID should differ from MQTT_TOPIC.

closed 2017-12-23 by alakdae