arendst/Tasmota #1419: PowStro Basic Support

camfarmer, 2017-12-23:
An ebay seller sent me some PowStro switches in place of the Sonoff Basic switches that I bought. 

https://www.dhgate.com/product/wholesale-powstro-smart-wifi-switch-ac-90/401010862.html

So I opened them up and discovered that they were using an esp8266 module. Then I found some information on the module and a datasheet.

https://fccid.io/2AKBPESP8266-S3
https://fccid.io/2AKBPESP8266-S3/User-Manual/User-Manual-3594791.iframe

I have added support for this PowStro switch to Sonoff-Tasmota. Is anyone interested in me submitting a pull request for this addition?

I did have to connect wires to the UART pins in order to initially flash these switches. It would be nice to figure out a way to use the SonOTA script to initially flash the PowStro switches.

davidelang, 2017-12-23:
Please submit a PR and also add them to the wiki.

hbg1979, 2018-01-02:
Hi, I also received a PowStro instead of a Sonoff. As I am not quite familiar with the development part and git (sorry for that and many thanks for all the good things you are doing) is the powstro support already available in the SonOTA update way?

pferona, 2018-01-03:
After some messing around, I was able to reprogram the PowStro with Tasmota. First you need to solder some wires to pins 3.3v, gnd, tx, rx. for reference, 3.3v is the top left pin. use the manual found at: https://fccid.io/2AKBPESP8266-S3/User-Manual/User-Manual-3594791.iframe for pin outs.

[image]

upload the usual way. 

once uploaded, you need to configure the pins since they are different than a normal sonoff. 

[powstro config.pdf](https://github.com/arendst/Sonoff-Tasmota/files/1599422/powstro.config.pdf)

GPIO4 is the relay
GPIO12 is the button
GPIO15 is a LED.

I hope this helps someone. 

Also, don't buy these on purpose, they are cheap and complicated... just stick with Sonoff!!!

camfarmer, 2018-01-03:
I have not been able to get SonOTA to work with the PowStro switches. Using
nmap, I have not been able to locate an open port. The next step would be
to try and observe the mobile app upgrading the firmware, if it even has
that capability.

I have only been able to program these switches using the hard wired method.

davidelang, 2018-01-03:
That's to be expected, SonOTA is hijacking the sonoff update mechanism to 
install tasmota on the device, PowStro will have it's own update mechanism and 
servers, so will be different than the sonoff one.

larrygardner1969, 2018-01-03:
I have everything connected as depicted in the above picture, and and it powers by the cp2102 converter, i can see tx and rx coming and going, but i keep getting 
espcomm_sync failed
espcomm_open failed
I have tried different  usb ports, different rates, and as much as i know up to this point.
1)  Is this a generic 8266 or a Wemos D1 mini as in pferona pdf ?
2)  How to identify if this thing is in flashmode?
3) Any different settings in the arduino sketch?  memory? and so on!
I KNOW i will never get these things again .. Tks for any help !

pferona, 2018-01-03:
Oh sprry,I forgot to mention when powering up, connect Gpio 0 to gnd with a jumper wire. When it is in flash mode, the lights will stay solid on.

larrygardner1969, 2018-01-04:
pferona -- Thank You very much .. That did it ...got it flashed and hooked up to my Domoticz Broker .. Your assistance is so very greatly appreciated ..

cookerrr, 2018-01-12:
I got one of these powstro switches when i really wanted a sonoff as well.   I have been trying to flash the firmware without success.  I grabbed the newest release, 5.11. attached pins just as described, shorted gpio0 to ground and all lights are lit up, I hit upload in arduino ide and get "warning: espcomm_sync failed     error: espcomm_open failed".  What am I missing?

pferona, 2018-01-12:
First make sure you do not have the tx and rx mixed up. ie:
rx --> tx
tx --> rx

also be sure your arduino settings are as followed. (these settings worked for me)
[image]

cookerrr, 2018-01-13:
I must have been to tired when i was messing with this.  I did in fact have the rx and tx reversed.  sadly when i was wiring it I wrote down the wire colors and what they were when i was plugging them in, but hence wrote them down wrong. lol.

Duesentr, 2018-01-13:
Hi there!

Thanks to all the help in this thread flashing the device was no problem.

I would also opt for activating the PowStro device configuration in the sonoff_template.h by default. I wrote my own configuration and just discovered someone already added one to the commented section at the end of the file after finishging mine ;-)

The PowStro actually offers two LEDs (one dual LED red & green).
green LED connected to gpio 15
red LED connected to gpio 13
The LED is connected to VCC directly, the negative pin pulled down by the esp module via a 10k resistor. The makes them "inverted", if I understand it correctly (even though I was not able to verify that, as I didn't find a command to see and set the gpio values in the console).

the other gpio are as gven above:
Switch gpio 12
Relay: gpio 4.

As the switch is not connected to gpio 0, entering the flash mode requires a jumper wire to "press the button" (pull gpio 0 to GND). The same holds true for the initial configuration, in case you did not flash a binary customized with your wifi setting. I did this right after flashing by using the 3.3V supply I used during flashing to power up the ESP, emulating the 4x button press with the jumper wire (as realized later that I had a typo in my SSID).

My device definition in sonoff_template.h looks like this:

 

> { "PowStro Basic",   // ESP8266, apparently sold as cheap replacement of the sonoff basic)
>      GPIO_USER,        // GPIO00 unconnected, Available on module
>      0,                // 
>      GPIO_USER,        // GPIO02 unconnected, Available on module
>      0,                //
>      GPIO_REL1,        // GPIO04 Power Relais
>      GPIO_USER,        // GPIO05 unconnected, Available on module
>      0, 0, 0, 0, 0, 0, // 
>      GPIO_KEY1,        // GPIO12 Button
>      GPIO_LED2,        // GPIO13 Red LED, active low (hence inverted?)
>      GPIO_USER,        // GPIO14 unconnected, Available on module
>      GPIO_LED1_INV,    // GPIO15 Green LED, active low (hence inverted?)
>      0, 0              //
>  }

Even though this might not be correct for the LEDs (either both should be inverted, or both normal), this gives me the following nice LED states:
Tasmota booted up relay off: green and red LED lit, though the red one is too dim and the overall appearance is green
Tasmota booted up relay on: green LED off, red LED on
Hence, setting the red LED to normal makes it "always on". Setting the red LED to inverted makes it "always off". Maybe the color achieved by my definition can also be achieved by setting both LEDs to inverted and tweaking the Tasmota settings, but I didn't dig deep enough for that yet.

The GPIO 0, 2, 5, 14 are unused but easily accessible on the module. Hence I set them to "user". They might be used for additional sensors.

**I would, however, only recommend sensors directly soldered to the PCB, with well insulated wires.** The power supply of the PowStro is a buck converter. Hence, **the "low voltage Vcc" is referenced to one of the mains input wires (either L or N, depending on how you wire the input)!**
The original sonoff, by contrast, uses a switching power supply. the low voltage VCC of the original sonoff is not referenced to mains.

The general PowStro PCB layout is not bad. It is fine, if you do not plan modifying/extending your wifi switch with additional sensors. As the price is not noticeably lower the the original sonoff, however, I follow the recommendation of pferona: do not buy it on purpose, go for the original!

Best regards

sejoki, 2018-01-13:
I also received the powstro switches and have been struggling with them for the last 3 hours… 

Is there any way to use sonOTA for these? They use a different WiFi (SmartLife-****) with no password, so sonOTA fails to recognize it.

It seems the only way to flash tasmota on the device is by the above mentioned soldering method, which wouldn't be the issue, but since I do not have a FTDI adapter at hand, I'm out of luck for now. It sucks that just because some idiot reseller wanted to save a couple of cents, I now have to spent more money to make them work for me.

Like it's been said already, go with the original, and don't buy them off eBay at all. Most of them don't even tell you they're sending you something different, some of them say they send out sonoff/powstro randomly and they're exactly the same (which is a lie) and others only sell the powstro, but at least they're honest about it. Either way, I haven't found a listing that reliably sells original sonoff switches.

Halo3000, 2018-01-16:
Hello
I have a problem and although after flashing I have no access to the Sonoff / SmartLife in the router, the device is displayed then I copy the IP would like to open the page and nothing is available.
what am I doing wrong?
the soldering worked
the flashing has worked
but something I have done wrong the sonoff.ino be changed except ssid and the key

larrygardner1969, 2018-01-16:
After flashing you can no longer use the Smart Life application.  Do you see the powstro in the client list of your router?  If you see the IP in the router then you should be able to access it. OR .. the web server of the powstro device is not working.  Did you put the SSID and password in the wifi area of the user_config.h tab and then recompile?
#define STA_SSID1              "your_ssid_goes_here"      // [Ssid1] Wifi SSID
#define STA_PASS1              "your_wpa_password_here"  // [Password1] Wifi password
#define STA_SSID2              "secondary_ssid if you have one"      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "secondary_password if you have one"  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect

larrygardner1969, 2018-01-16:
sejoki
The only way to be 100 % sure would be to order from the itead store
https://www.itead.cc/sonoff-wifi-wireless-switch.html

Halo3000, 2018-01-16:
yes, I typed everything as you just wrote.
after flashing, I see the Powstro in the IP list of my router. if I enter the ip in the browser only connection error. So not the usual tasmota surface.
Do I have to change other things?

larrygardner1969, 2018-01-16:
can you connect to the device by taking your computer and then using your wifi on your computer can you connect directly to the powtro?  If your flash was successful if i remember correctly the device webserver starts as dhcp 192.168.4.1 and your computer gets the ip of 192.168.4.2 .. you should be able to set it up from there.  If not here is the visual/written guide I used ...also if you look at the tasmota git there outlines some of the setup with the aruino ide:
https://www.youtube.com/watch?v=chyVjtYb0EA
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

larrygardner1969, 2018-01-16:
The latest version of Tasmota 5.11.1b if you use the github address .. does shorten steps from the video as alot of the libraries needed are included in the source file ..

jaybanane, 2018-01-16:
Hi,
I want to switch off the Leds of the powstro but I dont get it.
@Duesentr, you wrote that you got this. Can you perhaps send a screenshot of your configuration how you got the leds off?

camfarmer, 2018-01-17:
I have been able to disable the red and green LEDs by not specify the associated GPIO pin, but I have not figured out how to disable the blue LED. This LED is on the esp8266 module and I am guessing that it is tied to the relay GPIO pin (GPIO 4). Thus I don't think it is possible to disable the blue LED without physically removing it.

reloxx13, 2018-01-17:
[image]

Halo3000, 2018-01-17:
Hey guys
I've really tried everything but I just have no success.
What the hell am I doing wrong?
he is right to connect
the user_config is set correctly
the flashing works
I see the device on the router,
sometimes as Sonof70 .... sometimes as unknown with the same IP
copy the IP and press enter.
it loads after about 10 seconds timeout and there is no tasmota surface

jaybanane, 2018-01-17:
@halo3000 i dont do something different. I dont know how to help you here sorry 😢
@camfarmer: same here. But do you think that I could just solder it out? Wouldnt that have a bad influence on the circuit anyhow?

Halo3000, 2018-01-18:
I finally made it
my error was just a wrong board software had 2.4 installed, but needed 2.3;)

larrygardner1969, 2018-01-18:
Thanks for that feedback Halo3000 !

Magic01, 2018-03-01:
I have a Switch that is call "Smart Home" and works originally with a so calles Smart Life App (never tested) - the inside looks similar to this one here. And it is also working with the same settings for the GPIOs and so on.
Here are some pictures - maybe someone has the same one:
[image]
[image]

cpiekacz, 2018-03-01:
I also have problems with flashing the board. The light flashes solid green, connected TX<>TX and RX<->RX and get the following errors:

[image]

Should I modify the sonoff_template.h and add "PowStro Basic" section to the module settings? Anything more that should be done?

cpiekacz, 2018-03-01:
Ok, I've managed to do it. So, for the folks that will come here in the future: 
1. The TX and RX needs to be switched
2. You need to do the following changes in sonoff_template.h
a) add "PowStro Basic" section (see near the end of the file) to "const mytmplt kModules[MAXMODULE] PROGMEM = {" section
b) add POWSTRO_BASIC to SupportedModules
c) add POWSTRO_BASIC to "const uint8_t kNiceList[MAXMODULE] PROGMEM = {" section
3. Modify sonoff.h and set
#define MODULE                 POWSTRO_BASIC // [Module] Select default model

alfagero, 2018-03-20:
@ reloxx13 Your picture was very helpful. Thanks. But there must still be a bug in it(TXD-Pinout)

The pinout is the following

 schematic module:

GND ,TXD
--+------+------------+----------------+-------
3.3V RXD, unused for flash ,GIO Pin0

I connected it this way and connected the module (pressing gio0 on ground and module key. Then the LED lights up green constantly.

Flash with:
esptool -cf firmware.bin
or other possibilities.....

And it works!! yeah

Many thanks cpiekacz! Great help !

tjtjdgur2719, 2018-03-29:
I flashed it successfully, but I have an issue of the board getting disconnected after rebooting and I have to re-flash the board to make it work.. so I can not use the board any help?