arendst/Tasmota #1420: Sonoff POW - Measurement at Off state

Kniri, 2017-12-23:
A standard Sonoff POW is unable to measure the electrical power, when the relay is Off. If the measurement would work independent of the relay state, the Sonoff POW could be used as a simple master/slave switch, for example.

I would like to share my modification of a Sonoff POW. The objective was the ability to measure the power independent of the relay state. Fortunately the Sonoff got 2 electrical connectors which are not used by the Sonoff. They are marked with the letter „E“ and are intended to connect the protected earth of input and output. The Sonoff doesn’t use protective earth internally, so the connection could be made outside of the Sonoff. The “E“ connectors could be used for different purposes. I have used that connectors to lead the other relay contact out of the device (see the black wire on the left side of the photo). This connection makes power measurement possible, independent of the relay’s state.

For good documentation I’ve scratched off the black colour of the letter „E“ and modified it to „L“. Additionally I’ve labeled the output connectors “Switched“ and “Non Switched“ with a little sticker. That’s all on the hardware side.

The software needs a little modification, because the power will not be shown if the relay is in the OFF state.

I made two changes in the module **“xsns_03_hlw8012.ino“**

At Line 238:   if (hlw_cf_pulse_length  /* && (power &1) */  && !hlw_load_off) {

At Line 243:  if (hlw_cf1_voltage_pulse_length /* && (power &1) */) {

@Theo: It would be great, if you could take this into consideration in future versions of your code. Maybe you have an idea to make it configurable.

[image]

arendst, 2017-12-23:
I thought that the always on option would provide almost the same functionality...

Kniri, 2017-12-23:
In my opinion “Always On“ is not an alternative solution. Let me show you, how I am using the modified Sonoff. In an environment where real switches should work independent from home automation, this solution can help. 

Imagine, you have a corridor with 3 regular switches, to turn on/off the light from different positions. Without home automation you have to use an impuls switch (ELTAKO). 

It looks like this:
[image]

If you want to add home automation, but won’t loose the given functionality, the modified Sonoff POW is a good choice.

Then it looks like this:
[image]

With this solution you have the same functionality, even if your home automation system is not working. 
If your home automation system is working fine, you are able to determine if the light is on or off. Additionally the system can switch the light, (e.g. with PulseTime 5 and Power 1 command).

This is only 1 application. You can use the modified Sonoff POW also as a Master/Slave switch. This means that you can switch the relay depending on the amount of payload (for example: switch on a vent if power is greater than 100 Watt). Or for power savings (for example: switch off the Active Subwoofer if power is less than 20 Watt).

The advantage of a modified Sonoff POW is that the switching capability remains the same.

arendst, 2017-12-23:
Convinced.

majekw, 2018-01-10:
My use cases:

1. I have a bench with lot of things connected to mains. Also laptop with docking station and 5.1 speaker set. Speakers draw constantly about 10W.
My goal:
- measure power utilization of all these devices alltogether
- switch on speakers only when laptop is docked and active
- switch off speakers when laptop is put to sleep or undocked
- everything else should be powered all the time.

2. I want to measure my home server power draw, but I don't want ANY interruption in ANY case (firmware update, some error etc.), so simplest thing is to bypass relay. Now I have reliable power source and I can use relay for switching on/off other things like additional fan in hot days:-)

With this hack (thanks @Kniri) in every case I can do it with only one POW :-)

arendst, 2018-02-04:
There will be an option SetOption21 1 to always allow energy monitoring in version 5.11.1i

closed 2018-06-04 by ascillato