arendst/Tasmota #1422: New Alexa no more finding Wemo or Hue Devices

Zumili, 2017-12-23:
Hello folks,

as I run into problems setting up my Sonoff devices (Sonoff Basic) with Tasmota to get detected by Alexa, I found more and more informations on the net, that Amazon changed the protocol of how Alexa discovers devices. It seams, that e.g. Wemo devices only work with a skill anymore on the newer Alexa devices. The websever of the Sonoff's is working so I assum that they work as the should. 
But I was realy not able to find any of the Sonoffs with my Alexa devices.

So what hardware I used to test:
- A bunch of Sonoff Basics (1 week old)
- 2x Fire Tablett 7 (1 week old)
- 2x Fire TV Stick (2-3 months old)
- New Alexa App on different Smartphones 
- Fritzbox 7362SL
- WRT54GL with DD-WRT and OpenWrt
- WRT54G with DD-WRT and OpenWrt

So what I have tested:
- Flashed some different version of Tasmota onto the devices.
- Used 4 Sonoff devices
- Used different methods of setting up the IPs. Fixed IP and with DHCP
- Used different Alexa devices to try to detect the Sonoff devices.
- Used different router with different firmware where the Sonoffs and Alexa devices connect to

So at the moment I have no more ideas what else I can test.
I will be happy to get some Information on how I might can fix that issue.

Maybe there is also an update in process to fix that?

reloxx13, 2017-12-23:
UPNP and Selfopening Ports allowed on the fritzbox?

stredwizard, 2017-12-23:
Today I set up a Sonoff basic with tasmota, everything works as expected with the switches on gpio port and the v1 hue app finds and can control the so off.

Unfortunately Alexa won’t find the sonoff and I’d like to know if it’s something I have or haven’t done or is the same issue as above and amazon have changed something, if so is there a way around it? 

Thanks

jobongo, 2017-12-24:
I am having an issue with Alexa as well. Alexa finds the devices and can turn them on but when turning the device off, they come back on in a couple of seconds. Something did change in the last few days because everything was working a couple of days ago. It seems as though Alexa keeps trying to turn the devices on because it doesn't see them or gets any response from them as to what their state is.... Not sure but I know that fauxmo library made some changes recently and with that library, Alexa works ok....

stredwizard, 2017-12-24:
I’ve spent the last two days moving over to node-red, I had a rpi3 doing nothing so thought I’d give it a try, it seems to work well, annoying for me as it’s another link in the chain to malfunction, so if it does get fixed I’ll probably switch back.

chunter1, 2017-12-26:
My Sonoff S20, running the latest Tasmota firmware, does not get discovered by Alexa in Belkin WeMo emulation mode.
However in Hue Bridge emulation mode it gets discovered.

Zumili, 2017-12-26:
I also tried the Hue Bridge emulation but with absolutely no success. chunter1 have you made any special configuration or just set the network, the emulation and the friendly name to your needs and then it worked with Hue Bridge emulation? Can you post the configuration parameters you have changed in your configuration?

chunter1, 2017-12-26:
No special configurations from my side.
I only changed the ones you listed.

closed 2018-05-09 by stale[bot]