arendst/Tasmota #1424: Sonoff T1 2 CH

Majo96, 2017-12-23:
Dear Community,

is there a way to define the switches of the Sonoff T1 2 CH as pushbuttons?
I can only find the option to change the SwitchMode for external switches.

Would be nice if someone can help me.

Thank you!

davidelang, 2017-12-23:
you would need to change the module template to change the gpio from a button to 
a switch

jimmyw, 2018-01-06:
I dont think its supported, i started a wiki page with my findings here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1

SupraJames, 2018-01-06:
JimmyW, Coincidentally I got the same model of switch today and found your Wiki page. It made me worried, but my T1 2CH UK version (with RF) is working fine, so far.

Did you actually try flashing yours by bringing GPIO0 low during boot? I simply soldered a fly-lead from GND and held it to GPIO0 for a second or so after boot.

jimmyw, 2018-01-06:
No, i did not try that yet. I am trying to figure out how the esp is controlling the relays and reading the buttons first, so i know what gpios they use. It will be hard to find out after flashing tasmota. Im completely sure i can flash it taking GPIO0 low, but then what?

SupraJames, 2018-01-07:
I think you’re making it a lot harder for yourself. T1 2CH is already supported hardware and Tasmota knows how to do all that, so maybe you could look at the code.

I flashed mine yesterday and I verified that the ESP is properly recording the state of both relays even even when using touch controls.

geekypenguin, 2018-01-10:
I have tried flashing two 1ch t1's and while they flash fine they dont work after flashing (no wifi or manual control)

SupraJames, 2018-01-10:
Are you setting the correct parameters in the Arduino GUI as per the guide?

Getting any serial output if you boot up with the PC still connected? Just the touch board of course, not the AC part. 

I can’t imagine the 1CH version is vastly different.

geekypenguin, 2018-01-10:
Ive flashed using platformio which autodetects the settings.
Will try the arduino IDE next
A serial monitor just produces 1 line of gibberish on boot/reset.

geekypenguin, 2018-01-10:
Tried Arduino IDE but cant get it to flash.
What version of Tasmota you using?

geekypenguin, 2018-01-10:
Have now tried v 5.11.1a on both 1ch versions and now my 2ch version and none work

jimmyw, 2018-01-10:
I did some updates, and added a link to this issue from the wiki:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1

I had not have the time to test flashing the esp my self yet, but please update the wiki when its confirmed working.

SupraJames, 2018-01-10:
Does seem strange. After some initial stability issues both of my T1 2CH are now fine. Both relays work and both buttons are responsive.

@jimmyw you obviously spent a lot of time on the Wiki page and tracing circuits etc, but you never actually tried to flash Tasmota. Why is that? As far as I can see the T1 series have been supported for a while, they are defined in the source code and the GPIO details are known.

@geekypenguin I would suggest using Arduino with v2.4.0 of the ESP8266 core and try flashing with the released version of 5.1.11. It works for me.

geekypenguin, 2018-01-10:
Arduino IDE eventually fixed it i think. was also having issues with mcafee doing a real-time scan on the compiled files. 
Will try again with platformio with mcafee off tomorrow, and also test the 1 and 3 ch modules and update the wiki appropriately

jimmyw, 2018-01-10:
@SupraJames For the simple reason that i did not find any documentation how to flash it in the wiki, and i totally missed this issue. (i found another one where it seemed not to be working) Also i tried to figure out how to put it in download mode without soldering. When i found the configuration in the source code, i already wrote the wiki page.

vMeph, 2018-01-11:
is there any videos showing how to set ups the 2 chanel module?

geekypenguin, 2018-01-11:
still not having any luck with the 1 ch version, now on my third.
I either get the checksum error on boot or floods of fatal exception. 
Im going to try a new flashing tool when it arrives tomorrow and see if that gives me any more luck. Ive not tried the 2 ch again just in case i got lucky

webtop, 2018-01-11:
I just managed to flash a 3ch T1. Everything works after boot, and I've had it running 5.10 for the last hour with no issues (well, one issue, but that's for a topic of it's own).

I soldered a wire to GPIO0 and put the other end of it into the corner input from the back of the panel, the socket where it connects to the relay board. For me this was the top-left of the socket.
After I plugged in my USB TTL cable to my laptop, I waited 3 seconds and then remove the grounding wire from the socket.
I made sure that the board type was set to esp01_1m in my platformio.ini file. I also set my monitor_baud setting to 74880 in the ini file, although I'm not sure that had any impact.

For me there was no visible indication that I was in flash mode (my switch is actually this one: https://www.amazon.com/gp/product/B075F28DKH/ which has a T1 board in it).

geekypenguin, 2018-01-12:
Got myself a new (read bought rather than home made) flashing tool and I have successfully flashed all 3 versions of the t1 (1,2 and 3 ch variants). 
The only limitations I have found have been caused by a power supply issue, and the button push commands dont appear to work properly out of the box, until the module is configured as a t1.

I will update the wiki shortly with this info along with how to put it into flash mode. There is no easy way, the hidden button on the main board is just a reset button, but I used a jump wire to connect a ground pin on J2 to the IC side of R19 which was previously identified as being the GPIO-0

geekypenguin, 2018-01-12:
the plan is to replace all switches as the hue's i have at the moment have a low WAF as they have to be done via the app, using the  wall switch messes everything up.

Ive found the 1ch mode boots into flashing mode when pushing button 1 and the reset button. will double check the 2 and 3ch versions if they are the same

geekypenguin, 2018-01-12:
yep, entering flash mode can be done without modification, but the method changes based on the version.
I am focusing on getting the last of my units flashed today and will update the wiki tomorrow with my findings.
I think its safe to say tasmota support on the t1 has been proven?

calcipher49, 2018-01-12:
Have you tried OTA flashing (using Sonota)? if it works it can be a good alternative without any hardware manipulation...

geekypenguin, 2018-01-13:
From the units I have, they all came with a version of the stock FW which is not compatible with SonOTA. No hardware manipulation is required to flash the T1s, with the exception of hooking up a serial programmer, which can be done with jumpleads and no soldering etc. I am currently writing an update to the wiki which has full instructions

marrold, 2018-01-14:
Apologies for potentially hijacking this thread, but the following is not clear from the Wiki - 

Is it possible to publish switch presses via MQTT?
Is it possible to publish switch presses via MQTT _without_ triggering the relays?
Is it possible to switch the relays via MQTT? 

Thanks in advance

davidelang, 2018-01-14:
yes
yes
yes

lookup switchtopic for the first two, the third is the common mode of operation

On Sun, 14 Jan 2018, Matthew Harrold wrote:

> Date: Sun, 14 Jan 2018 01:07:15 +0000 (UTC)
> From: Matthew Harrold <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f31f64473a3505987c3d049be67db8fb5fe70fa0b92cf0000000116726f4
>     392a169ce10f217db@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Sonoff T1 2 CH (#1424)
> 
> Apologies for potentially hijacking this thread, but the following is not clear from the Wiki -
>
> Is it possible to publish switch presses via MQTT?
> Is it possible to publish switch presses via MQTT _without_ triggering the relays?
> Is it possible to switch the relays via MQTT?
>
> Thanks in advance
>
>

gheesung, 2018-01-14:
I am reading the instructions from the wiki. Does that mean that the GPIO 0 is no longer need to be connected to ground (as mentioned in the other threads)?

@geekypenguin I am trying to flash mode using the instructions ie press down button4 and button1 a few sec after boot. The blue led shows 2 short brinks consecutively but still cannot flash. Any hints? Thanks.

geekypenguin, 2018-01-14:
Yes you no longer need to solder to gpio0 to connect to ground as it can be achieved using the mentioned combination or button presses.

Wait for the so off to boot, then hold button 1, press and release 4, count to 3 and release 1. Should now be in flash mode. The main led will be off and the LEDs behind the touch buttons will be on faintly

geekypenguin, 2018-01-14:
I'll update the wiki if it's not clear

gheesung, 2018-01-14:
@geekypenguin I have managed to get to the "blue tint" mode using the steps you have mentioned but still cannot flash. Are you using esptool to flash and what are the flashing parameters? Thanks.

geekypenguin, 2018-01-15:
I have been using Arduino IDE to flash. 
What is your output of the above? Is it showing a connection but not flashing, timing out before connecting,  or flashing but then not booting?

geekypenguin, 2018-01-15:
The wiki has the parameters for using esptool as
esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 sonoff.bin

gheesung, 2018-01-15:
The serial link is connected but not flashing. The behaviour is similar to the behaviour when the ESP is not put into flashing mode .

I am using prebuild bin file because the arduino compiled version is greater than 512k and caused the ESP to reboot in my sonoff touch. 

I have 4 new T1s waiting to deploy to my smart home. I have an earlier batch which is able to flash via Sonota but this batch the SSID is not broadcasted. Feeling desperate now.

geekypenguin, 2018-01-15:
What does a serial monitor at 74880 baud show?

gheesung, 2018-01-15:
I think there is a labelling error in your diagram. the pin beside the 3v3 should be RX.

I managed to flash but the settings keep reset back to default. e.g. when I set my ssid to my home ssid, it reset back to "indebuurt1" when the ESP restart itself.

geekypenguin, 2018-01-15:
It isn't my diagram but yes, there have been discussions about the labeling and the conclusion was that it depends how your programmer is labeled. Some are labeled as to where to connect (Rx to Rx for example) and others are what they are (Rx to tx)

That is another reason to roll your own with credentials baked in. When the so off resets a few times in quick succession it resets to default configuration. It will reset frequently if powered by your programmer as they rarely have the capacity

gheesung, 2018-01-15:
Just to share my experience after successfully flashed 4 set of T1s. 

1. I think the diagram is not correct as the marking is on the T1 board and it should show the Pinout as shown on the board, not on the programmer pinout.
2. Flashing the T1 only needs to press button 4 (long press ~2-3s). Pressing 1 is not required. If it doesn't go into flashing mode, do again. you should see something like when the serial port baud rate is set to 74880.

 ets Jan   8 2013,rst cause:2, boot mode:(1,6)

when this is shown, the T1 is ready to flash with the Tasmota firmware.

3. I have the problem of flashing my own compiled version. It keeps rebooting after I have uploaded. For the pre-built firmware, using it has a chicken and egg problem. Because there is not enough power to drive the board, the setting keep reseting to default. Connecting the T1 to main while my FTDI is connected is very risky. The alternative method which I am using to prevent electrocuting myself is to setup a guest wifi "indebuurt1" at my router. I powered up the T1 and reconfigure it to my network.

Hopefully, this will help others who are planning to use T1. Really missed the sonota update method. I think iTead has stopped boardcasting the SSID. I have a new batch of S20 and POW also have the SSID hidden.

geekypenguin, 2018-01-16:
Yes, a long repeated presses does sometimes drop you into programming mode as it upsets the bootloader thinking there is a problem with the sketch so drops out to programming mode, but this is not the correct or reliable way to enter the bootloader

kylegordon, 2018-01-19:
For some reason I am getting similar behaviour as @gheesung with my T1 2CH- where the device can be mostly configured (connected to an AP and MQTT broker) and then for no reason it will reboot and lose all of its configuration.

I am using release 5.11.1, flashed successfully with esptool.py. IP configuration is either done with SmartConfig or WifiMaanger - either way, and it will seemingly randomly reboot either mid configuration, or after a convincingly long period of communication with MQTT.

It's not a power issue, as I'm measuring the current draw through USB and the UART, and it's pulling 0.1A - 'traditional' USB is good for 0.5A

In addition, frequently after multiple resets, the Tasmota SSID stops being available, and the device has to be hard power cycled.

kylegordon, 2018-01-19:
Plugged it into the relay board and mains, and it worked for 5 minutes. When sitting idle for a while, it beeped once, and again twice, and then reset. It toggled off relay/switch 2, and now appears to be power cycling whilst in single relay/switch mode. Button 2 is dead to the world, and button 1 is only available for short periods of time (presumably between reboot loops).

Understandably it's unusable in this state

vMeph, 2018-01-20:
using the 2 chanel module with mqtt i doenst have state topic only control topic, any way to have a state topic?

gheesung, 2018-01-20:
@kylegordon, Tasmota is not able to save the setting when it is powered by the FTDI, the firmware is reset back to the default setting which is Sonoff basic (if you use the pre-compiled bin file). I have not tried long usage and will find some time this weekend to check the stability and post my finding.

geekypenguin, 2018-01-20:
@kylegordon: when it power cycles like that it does default back to basic settings which is why it looses connection, and everything but button 1 becomes unresponsive.

Of the 15 I have flashed I had two do this to me. One was solved with by erasing the flash then a re-flash but unfortunately the other couldn't be saved. I don't think it is an issue with the tasmota firmware but the boards themselves. (Swapping the parts around, the issue follows the front board with the esp). 
My only option in this case was to return it for a replacement unit.

@vMeph. Please expand on what you are meaning? What are you trying to change? Some settings can only be changed via the console (or via mqtt instructions) and not the UI.

kylegordon, 2018-01-20:
@gheesung Thanks for the info. I'd been flashing it and configuring IP and MQTT details whilst on FTDI, and then transferring it to the mains powered unit in order to change the module configuration and test it. 

Later today I'll erase it, reflash it, and then immediately do the rest of the config on the mains unit, and will report back

kylegordon, 2018-01-20:
@gheesung @geekypenguin Can confirm, you guys are right. It will only operate in a stable fashion when attached to the mains backend. The ESP82685 board doesn't draw that much current, so it must be failing to access a chip on the relay board and then power cycling, or something.

Problem solved, though. Thank you :+1:

vMeph, 2018-01-20:
@geekypenguin  got it to work mate :)

vMeph, 2018-01-21:
@geekypenguin im having a problem with the touch t1, when i touch 1 button to close then he will turn back on, or will turn the other on, what this could be, this dont happens all the time but it happens

vMeph, 2018-01-22:
i figuer ou the problem, its a balast light i got pulggen into the son off witch produces a eletromagnetic field that will trigger sonoff t1, i changed the light to a normal one and wotks perfect

vMeph, 2018-01-22:
i wasnt able yet to upload firmware with OTA, any one could give info with the required steps to be able to flash with OTA

geekypenguin, 2018-01-22:
@vMeph Flashing Via SonOTA isnt possible any more. If you are referring to flashing a new version of Tasmota over an existing one then it is the same as for any other Sonoff device

kenmcc, 2018-02-02:
FWIW I was able to reflash tasmota over my v1.6 stock firmware last night, following the instructions on the web page (hold button 1, then button 4 for 4 secs or so, release button 1, release button 4). Hardest part was maintaining good contact with the pads, I didn't solder in any headers as not sure the clearance is there for them to be there permanently. Might look at sourcing pogo pins if need to do several.

angelpereyra, 2018-02-21:
Hi all, which is the configuration needed on the Arduino IDE? ESP8285, 74880 baud,... ?

Thanks!

geekypenguin, 2018-02-22:
See the wiki...

FatBeard, 2018-03-02:
Hi guys. Is anyone having problems with the T1 UK 1 gang and 2 gang turning on and off randomly without being pressed. Button 1 on both variants is causing problems. Button 2 seems fine. Something appears to be setting off the capacitive button post Tasmota flash.

SupraJames, 2018-03-02:
I have 4 of these (2 and 3 gang) deployed and have never had that happen with any version of the firmware.

jeylites, 2018-03-02:
@FatBeard 
@SupraJames 

Flash with ESP tool followed by Tasmota. 99% of the time issue is fixed.

FatBeard, 2018-03-02:
Thanks for the response guys. @jeylites Are you saying to flash with the esptool first and then flash tasmota over the air? Also have you had the issue i described or is esptool/tasmota just a way of fixing issues genericall?

jeylites, 2018-03-02:
@FatBeard 

Yes Flash with ESP Tool as shown below and on the third attempt flash with Tasmota. Out of twenty eight , 3gang T1s that I flashed, only three had an issue like yours. So much so these days any new Sonoff, I clear out the flash before re flashing with Tasmota simply because I do t want to do it twice.

Over the air something I have yet to explore. All my flash has been manual.  I got to learn how to do OTA at some point. 

> Open a command prompt and execute command
esptool.py --port COM5 erase_region 0x0F4000 0x008000
Optional Clear the complete flash with command
esptool.py --port COM5 erase_flash

jeylites, 2018-03-02:
@FatBeard is your UK t1 the latest version or old type. Old time meaning PSF module types...

I have both OLD and NEW, it bothers me that the  back Light on the NEWER ones  are dimmer than the OLDER  one which makes it difficult to see the buttons at night. 

Do you have similar issues?

FatBeard, 2018-03-02:
@jeylites I seem to have the older ones with the PSF sticker/module. Didn't realise until now that there were different versions of it. See attached below.

[image]
[image]

jeylites, 2018-03-02:
Yeah, that’s the old one. Thinking of it I like the old one better for its
bright LED touch :)

I currently looking at way to crank up the brightness of the newer ones. If
any one have ideas, post it up...

MTrab, 2018-03-02:
Have any of you tried this version of a T1 EU 2 Gang?
Can't see any holes for the pins to flash the MCU.
[image]
[image]
[image]

[47 more replies]

closed 2018-08-25 by ascillato2