arendst/Tasmota #1425: Offset BMP280 sensor

JPBreestraat, 2017-12-23:
I installed a bmp280 next to a dht11. The output of the BMP280 sensor is a bit offset.
[image]

is there a way to correct it? ( for the dutch in jip en janneke taal)

thank you

Frogmore42, 2017-12-23:
What elevation are you at?
With a pressure reading of 565 hPa, you are either at a very high elevation or have something very wrong.  Similarly, a temperature of -142.2 is probably not just a little offset, it is very wrong.  That either says things are hooked up incorrectly, you have a bad part, or something is wrong with the code.  I don't have a BMP280 so can't say which of those it might be.

JPBreestraat, 2017-12-23:
the elevation is +3m, i also have a bmp180 and it's also offset. not that high as the bmp280 but it stil is(70C)

Frogmore42, 2017-12-24:
It sounds like there is something significantly wrong with your setup, or your BMP280.  I discovered that I had a BMP280 in my parts bin.  So, I hooked it up and it is reading numbers that appear to be correct. 

[image]

So, it looks like it is possible to get this working.  When I connected it to a NodeMCU device, it did not work at all, but that device has had issues in the past (not sure if it is the device or the breadboard that it is on).  I did not have to add any pullups on the I2C lines.

JPBreestraat, 2017-12-24:
Eureka, i've solved the problem.........a bad wire with too much resistance (i think).

closed 2017-12-24 by JPBreestraat