arendst/Tasmota #1426: GPIO14 Basic Push button operation

DaveTiff, 2017-12-23:
Hi
I want GPIO14 (Pin 5 on the header) to act the same as the built in push switch but without the issue of somebody pushing the switch for a long time and setting things to default.

I have change line line below to Pushbutton, but it behaves as Pushbutton_toggle (Wall Switch)
#define SWITCH_MODE            PUSHBUTTON

I want a Toggle action on GPIO14, what am I doing wrong.

I have tried all the options on the web interface?

Regards Dave

reloxx13, 2017-12-23:
if u want a toggle, why did u change to pushbutton?...
a pushbutton is the bell at your door, dont ringing after releasing the button.

stay on toggle

be sure the new user_config is loaded by changing CFG_HOLDER

web interface:
config > modul config > GPIO14 = 09 Switch 01

done

DaveTiff, 2017-12-23:
Many thanks for the reply
I have tried that but "config > modul config > GPIO14 = 09 Switch 01" sets it working as a wall switch, i.e. in toggles on change of state, rather than a momentary switch?

Re Dave

reloxx13, 2017-12-23:
okay, let it stay on 09 switch and enter in the console:

savedata 1
switchmode 3
savedata 0

PUSHBUTTON would be right then.

DaveTiff, 2017-12-23:
Hi That works perfect, thank you

Why cant I set that in the user config file?

Or is it because I did not know about CFG_HOLDER.

What do I need to set in my  user config file? as I have another one to upload?

Re Dave

reloxx13, 2017-12-23:
did u set gpio14 to 09 switch before in ur first tries? 
u canot set the gpio14 default in the user_config.

it could be the CFG_HOLDER, without changing it, the new config does not get loaded or you did not set the gpio14 to switch mode.

Both settings needs to be done, change SWITCH MODE to PUSHBUTTON (change CFG_HOLDER to a new vlaue to load this new config) and set the GPIO14 in the web interface to 09 Switch.

DaveTiff, 2017-12-24:
Hi reloxx13

Yes that did it.
The CFG_HOLDER can it be changed to any number? or just incremented by 1?

many thanks

davidelang, 2017-12-24:
the only thing with CFG_HOLDER is that it needs to be different than the prior 
value.

using the date works very well

closed 2017-12-27 by DaveTiff