arendst/Tasmota #1428: Rf bridge and DW1

seicento, 2017-12-23:
Hi.
I have flashed fw on the bridge and connected 6 dw1 sensors to it. My problem is now that i cant figure out how to make them binary sensors i Home Assistant?
I cant seem to find any information on it anywhere.  Need more onfo on the mqtt topics for every sensor

Thanks

Reikard80, 2018-01-04:
Hi All,

I have same problem.. Does anyone can support to provide the code to put in configuration.yaml ?

I have a Sonoff RF briged flashed with Tasmota and I have managed to learn some RF signals from Sonoff Sensors.. CT60-Wireless and CD100S-Wireless Door Detector.. Mqtt mosquito is running well and firmware in RF bridge has mqtt enabled.. I really just need the the binary code to put in configuration.yaml

So far i got faillure by putting this one.. :(

binary_sensor:
  - platform: mqtt
    state_topic: "stat/bridgeRF/rfKey4"
    name: Door sensor
    qos: 0
    payload_on: "1"
    payload_off: "0"

Waiting your suggestions

Rik

seicento, 2018-01-09:
gave up on the sonoff bridge. Dont think that anyone can git it to work. Have now ordered xiaomi gateway and door sensors

Reikard80, 2018-01-11:
Ok but can I use xiaomi gateway to clone a RF signal a use it for another product? Or you can just use xiaomi products with this gateway? 
I need to clone may blinds remote control and add it on home assistant.. therefore a bridge like sonoff was perfect

mr-sneezy, 2018-03-18:
This stuff with binary sensors like RF door or PIR devices is cause all sorts of bother for many people it seems (and I though it was just me for while but no). I also have the Sonoff RF bridge, now with Tasmota FW. This guy has spent so much effort on the FW it's a pity we can't quite make the last leap to get use out of it in HA
Perhaps if we put up the right info we can get a usable answer from somebody more knowledgeable (which is what forums were for).

I setup my Sonoff bridge with Tasmota, 'learned' a couple of RF devices and then captured the MQTT packet going to HA when I triggered them. What I need to be shown is how to construct a sensor entity for rfkeys in my configuration.yaml. Once I have that I think I can run with it. Even better would be a binary sensor entity that resets to off after a few seconds (because the RF device like PIR won't report the OFF for it's self).

Here's what the tasmota FW MQTT message looks like, captured with MQTT Snooper. I'm not sure what to do with the JSON payload to make a binary switch of it....

 tele/sonoff_bridge/RESULT
 {
  "RfReceived":{
   "RfKey":1,
   "Data":"3DD105",
   "Low":160,
   "High":480,
   "Sync":5040
 }
}

mr-sneezy, 2018-03-18:
Got a tip from another thread. This works to read the sensor for RfKey1, for RfKey2 we'd just change the payload_on string to "2". The payload_off does nothing here though.

```
binary_sensor:
  - platform: mqtt
    name: "Test RF bridge rfkey 1"
    payload_on: "1"
    payload_off: "0"
    device_class: opening
    state_topic: "tele/sonoff_bridge/RESULT"
    value_template: '{{ value_json.RfReceived.RfKey }}'
```

ascillato, 2018-06-09:
Thanks. Added to the wiki :+1:

closed 2018-06-13 by ascillato2