arendst/Tasmota #1429: WeMo devices issues with Alexa

jobongo, 2017-12-24:
First of all I have to say that this is a great firmware and thanks for the work. I am having an issue when connecting my devices to Amazon Alexa using WeMo emulation. The devices are discovered and they can be turned on but when turning of the Sonoff through the Alexa App or through the Echo, after a couple of seconds, the relay turns back on again. The Alexa app reports the device as "Unresponsive."

I had the same issue when using Fauxmo libraries and I think that Amazon changed something in the way that it handle WeMo devices. From what I understand, the echo is now expecting a status message from the WeMo device to say whether it is on or off. The fauxmo libraries made a change to do this and this problem went away.  I have been looking through the code to see if I could implement the change myself, but I'm not sure exactly where it is handled. Which library is being used to emulate WeMo devices?

Also, I have tried disabling MQTT (Same issue..) and when turning a WeMo device back on, through Alexa, the devices keeps publishing its status to ON (See attached).
[image]

reloxx13, 2017-12-24:
use latest dev build
https://github.com/arendst/Sonoff-Tasmota/pull/1357

jobongo, 2017-12-25:
I downloaded and compiled from the latest devel build and it seems to work but I am still getting the same error with the alexa app say that the device is unresponsive. The lights turn off and stay off.... I was using build v5.10.0a and that fix, if I'm not mistaken, was already implemented in that version. Thanks for the help.

closed 2017-12-25 by jobongo