arendst/Tasmota #1430: Help parsing sonoff json in python

foxsam21, 2017-12-24:
I am looking for some help if there is anyone who can direct me in python to write a script to parse data out of the sonoff's mqtt messages. For example I want python to store IPAddress as a variable but it comes as different topics on startup or status command. 
also I want python to receive temp from sonoff sensors and upload them to influxdb.

If there is anyone who can point me out in the right direction that's great.

Frogmore42, 2017-12-25:
What are you running the Python on?  Why not use Node-Red?

foxsam21, 2017-12-25:
I don't know anything about node-red. I will try to look for some examples
and see if I can get it working that way...

closed 2018-05-08 by stale[bot]

amotl, 2019-06-06:
Dear @foxsam21,

we recently have been able to dedicate some time to implement an appropriate decoder [1] for the [Kotori data historian](https://getkotori.org/). You might be able to reuse what we have been building there either by running Kotori itself [1,2] if you are into metrics acquisition or by reusing code from [3].

With kind regards,
Andreas.

[1] https://getkotori.org/docs/decoders/
[2] https://getkotori.org/docs/decoders/sonoff-tasmota.html
[3] https://github.com/daq-tools/kotori/blob/master/kotori/daq/decoder/tasmota.py