arendst/Tasmota #1432: Sonoff basic with  DS18B20 , temperature adjust

sp2003, 2017-12-24:
it's possible to adjust temperature?
I need to add 2°C to detected temperature

ionciubotaru, 2017-12-25:
Do this in you HA, there is no reason to "adjust" temperature in Tasmota

sp2003, 2017-12-25:
merry xmas!

not exactly, because i use sonoff's temperature in domoticz and other system. if sensor is modifiable, all systems is ok

l1q1d, 2018-02-03:
I agree with sp2003, an additional command to adjust the temperature offset could be really useful.
I see that different sensor even in the same room show different temperature.

fennec622, 2018-06-02:
Yes please possible adjust temperature
domoticz adjust value don't work with mqtt domoticz/in

ascillato2, 2018-07-10:
Hi,

You can use rules for adjusting temperature and then publishing to a topic.

See https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#8-arithmetic-commands-to-be-used-with-vars

Thanks

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2