arendst/Tasmota #1433: Use of a CO2 sensor MH-Z19b

FabianHM, 2017-12-25:
With great interest I have read that the MH-Z19b is now also supported by Tasmota. I have currently two questions:
**Which pins of the device are needed, probably Vin; GND; TX. Is that correct?**
[image]

The module can not be installed in a housing because of the sensors. 
**Does anyone know a suitable housing, where above the board the module body could be freely installed?**

Merry Christmas 
Fabian

arendst, 2017-12-25:
You'll need Vin (5V!), Gnd, Rx and Tx.

[image]

FabianHM, 2017-12-26:
Will the Sonoff boards at VCC supply the required 5 volts for the sensor? Is this "Wemos7 V2 in its specifications equal to the MH-Z19b or is there any new dialogue with the new sensor planed?

FabianHM, 2018-01-12:
After I get my WeMos D1 and the MH-Z19B.
I configured it as described, but I dont't get correct Informations.
[image]

How can I check if it is a defective sensor or where the problem is?

arendst, 2018-01-12:
Pls show the modules parameters page

FabianHM, 2018-01-12:
I solved the problem
In the example RX and TX were reversed.
Correct is:
[image]

FabianHM, 2018-01-12:
The display of CO2 works fine, but the temperature will alway show in whole numbers, no decimal places.
Please you can check this?

spencer-at, 2018-02-05:
hi, 

can I use the MHZ19B sensor with a basic Sonoff or do I need a different hardware to run it?