arendst/Tasmota #1434: Hue emulation works with Alexa but she says,

halpz, 2017-12-25:


Frogmore42, 2017-12-25:
I use ha-bridge and have it configured to send mqtt commands to turn on/off things.  I did this because I had a light that was not hue capable but did support mqtt (with another gateway).  I have not tried the emulation with Tasmota.

Frogmore42, 2017-12-25:
Looks like issue 1429 is similar and there is a fix there.  Or, you can use ha-bridge like I do.

chunter1, 2017-12-26:
Got the exact same issue with version 5.10.0c

arendst, 2017-12-27:
Pls try again with 5.10.0d and report progress.

halpz, 2017-12-29:
issue remains on 5.10.0f

KoLSMS, 2018-01-04:
Hello, same problem Here... :(

geekypenguin, 2018-01-15:
Same issue on 5.11.1, the devices add no problem and Alexa app shows the status but when sending a command it says not responding

KennethLavrsen, 2018-01-17:
I have same issue. 5.11.1b. Alexa can discover and show the device in Alexa app but it cannot control it. Alexa says that the device does not respond and nothing happens when to try and turn on or off. The app shows the correct on or off state of the device so it seems that Tasmota ignores the commands from Alexa

TekMason, 2018-01-17:
Using Hue Bridge emulation on 5.11.1 the sonoff basic works as it should for me with one exception:
When I say, "Alexa, light on" the light turns on but Alexa replies with "light isn't responding".

KennethLavrsen, 2018-01-20:
See my comment on what I think is a duplicate of this  https://github.com/arendst/Sonoff-Tasmota/issues/1639

OFreddy, 2018-01-24:
See my last comment in #1639, this might also fix this issue.

reloxx13, 2018-01-24:
Heya @arendst ,

it seems like alex ais not sending args to the webserver anymore when changing the state.

i made some debug outputs in the code and received this:

21:33:37 HTP: Hue API (/api/dIUhc67uzIHamo5mYqfQVjkjgRMsvkXeZo4u4Fg/lights/1)   
21:33:37 HTP: Hue POST args (NO ARGS)   
21:33:37 HTP: Hue API (/api/dIUhc67uzIHamo5mYqfQVjkjgRMsvkXeZo4u4Fg/lights/1/state)   
21:33:37 HTP: Hue POST args (NO ARGS)   
21:33:37 HTP: Hue API **E2** ()   
21:33:39 HTP: Hue API (/api/dIUhc67uzIHamo5mYqfQVjkjgRMsvkXeZo4u4Fg/lights/1/state)   
21:33:39 HTP: Hue POST args (NO ARGS)   
21:33:39 HTP: Hue API **E2** ()   
21:33:43 HTP: Hue API (/api/dIUhc67uzIHamo5mYqfQVjkjgRMsvkXeZo4u4Fg/lights/1)   
21:33:43 HTP: Hue POST args (NO ARGS)   
   
this is state switch on/off from alexa app
/api/dIUhc67uzIHamo5mYqfQVjkjgRMsvkXeZo4u4Fg/lights/1/state
   
this is request state of device (repeated while device open in alexa app)
/api/dIUhc67uzIHamo5mYqfQVjkjgRMsvkXeZo4u4Fg/lights/1
  
   
   
[image]
[image]

E2
[image]

i could not find out why the args are missing.

closed 2018-06-08 by stale[bot]

koraysels, 2018-12-11:
issue is still here v5.12.0

reloxx13, 2018-12-11:
we are on 6.3.0.x

koraysels, 2018-12-11:
ha wow the OTA didn;t work then ... lets try that again !