arendst/Tasmota #1436: Will not connect to wifi. 4 press wifi not found from computer

eirikgund, 2017-12-26:
Hi. I have flashed the standard sonoff and serial monitor from arduino ide gives me:

00:00:07 WIF: Connecting to AP1 error2,4 in mode 11N as sonoff-3346...
00:00:15 WIF: Connect failed as AP cannot be reached

As you see my ssid contains a comma in the middle. Is this a problem?

Next issue is. When I press 4 times it says:
00:00:14 WIF: WifiManageractive for 3 minutes
00:00:15 HTP: Web server active on sonoff-3346 with IP address 192.168.4.1

But when I check to see if a new ssid is available there is none. I cant find it. Not on my computer or on my phone. What is wrong?

gaemsli, 2017-12-26:
I have several Sonoff Basic, POW, TH16, 4CH running with firmware 5.9.1 without any problem. Today i tried 5.10.0 on the new arrived Sonoff Bridge and ist working great too. So i started to flash 4 Sonoff Basic with the firmware 5.10.0 via uploading binary (Webpage). BANG!! All off the new flashed sonoff where connecting to the wifi, get a DHCP IP-Adress and mqtt but only for a short time and then they restartet.
The 4-time-button-press give me the sonoff-wifi-ap but i am not able to connect to that. no way to communicate via mqtt or browser. Because they are in production i had to reflash the Sonoff basics via serial upload with the firmware 5.9.1. After that painful downgrade the work as they done before.

Interresting: the bridge do not show that behavior...
i have not testet other devices because i have no spare ones at the time...

It seems that something is wrong with that release for sonoff basic ...

eirikgund, 2017-12-26:
I downgraded to 5.9.1 and now it connects to my wifi but I am still not able to reach the web server bu typing in the ip iddress in my browser. But progress is progress ;)

closed 2018-05-09 by stale[bot]