arendst/Tasmota #1439: T1 UK3C Troubles entering

Swarley321, 2017-12-26:
Hello,

I am new to the Tasmota community and I am having basic problems with my new T1 3CUK. I apparently have a clock on board that needs another baudrate than 115200 (74800?). Which modifications to the Tasmota code do I have to make in order to make sure that the data transfer is done with the alternative baud rate? 
I am also not sure if I am putting the device into flash mode corretly. When I connect the device to a standard terminal and set the baud rate to 74800, I get different boot modes with every reset (whilst holding GPIO0 to GND). Has anybody experienced issues like this on such a board? 

Below you can find pics of my setup. Thank you!

[image]
[image]

Swarley321, 2018-01-02:
Hope anybody can help, since I a want to use the Sonoff touch very soon. Thank you very much!

Jim1948, 2018-01-02:
You have GPIO-10 marked on photo. GPIO-0 is third from left in photo.

Jim1948, 2018-01-02:
[image]

Swarley321, 2018-01-04:
Hi Jim, thanks for the tip, I marked it wrong on the pic but did it actually right. Meanwhile I tried the OTA method, which also went wrong somehow, a mysterious Eception 9 is displayed on the terminal. I just also found out what the problem was with flashing via terminal and PlatformIO, the FTDI chip has a small production error, the RX pin was slightly welded together with one of the control pins. However, now that I have a failed OTA flash, I asume that it also is impossible to flash the device via FTDI chip. I keep trying but it does not work. Is the device ready for the trash can now?

Swarley321, 2018-01-04:
This is the excetion text followed by the stack content: 

Exception (9):	epc1=0x401051fc epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000b depc=0x00000000	

ctx: sys 	
sp: 3ffff800 end: 3fffffb0 offset: 01a0	
	
>>>stack>>>	
3ffff9a0:  402b25a0 40294bcc 00000003 000fc000  	
3ffff9b0:  4029a2e9 3ffee1de 3ffee1de 4029a2a7  	
3ffff9c0:  4029a920 60000e00 3fff16cc 4029a8fb  	
3ffff9d0:  401050e4 000000fd 000000f7 60000200  	
3ffff9e0:  3ffffedc 4e5d486b f8cb7f4b cb3349bc  	
3ffff9f0:  ffffe6fd ffffffff ffff0003 00000000  	
3ffffa00:  00000000 00000000 00000000 00000000  	
3ffffa10:  00000000 00000000 00000000 00000000  	
3ffffa20:  ffffffff 00ffffff 00000000 00000000  	
3ffffa30:  00000000 00000000 00000000 00000000  	
3ffffa40:  00000000 00000000 00000000 00000000  	
3ffffa50:  00000000 00000000 00000000 00000000  	
3ffffa60:  00000000 ff000000 ffffffff ffffffff  	
3ffffa70:  ffffffff ffffffff ffffffff ffffffff  	
3ffffa80:  ffffffff ffffffff ffffffff ffffffff  	
3ffffa90:  ffffffff ffffffff ffff00ff ffffffff  	
3ffffaa0:  0000000a 616e6946 6174536c 3f006567  	
3ffffab0:  3ffefd94 3ffe869b 85940162 3fffa956 

...etc,

if this helps anybody to understand my problem.

Swarley321, 2018-01-04:
This is the Terminal result when I switch from 74880 baud to 115200. The baudrate seems to be 115200 in boot up, but when it switches to the application it seems to switch to 74880. Boot mode is (3,6) when I connect with the GPIO0 Pin at low, is this the correct boot mode for flashing?

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)	
	
load 0x40100000, len 2408, room 16 	
tail 8	
chksum 0xe5	
load 0x3ffe8000, len 776, room 0 	
tail 8	
chksum 0x84	
load 0x3ffe8310, len 632, room 0 	
tail 8	
chksum 0xd8	
csum 0xd8	
	
2nd boot version : 1.6	
  SPI Speed      : 40MHz	
  SPI Mode       : DOUT	
  SPI Flash Size & Map: 8Mbit(512KB+512KB)	
jump to run user2 @ 81000

Swarley321, 2018-01-05:
Hello,
I managed to flash via Wire finally. It turns out that it is necessary to be in the boot mode 1 (GPIO 0 to GND, GPIO 2 to VCC (3.3V), GPIO 15 to GND) in order to flash correctly. Since this is a 74880 baud device, it is necessary to set the:
; *** Serial Monitor options
monitor_baud = 74880
in the platformio.ini file of the tasmota master 

Following this turtorial (except for the wiring), helped me to manage the flash:
https://www.youtube.com/watch?v=n4MDRm2yAJg

The pic below shows the correct pin connection for boot mode 1 for the T1UK 3CH. 

Thanks a lot and BR

[image]

closed 2018-01-05 by Swarley321

joba-1, 2018-11-24:
Hi @Swarley321,
thanks for sharing this - especially the picture. I already started to think  I am the only one with this thing :)

FYI or anyone else trying this in the future: it was not necessary for me to connect gpio 2 or 15. They usually already have the correct voltage (see your table). Only gpio0 to gnd during S1 press was enough.

I have one issue though (using version 6.3.0): I cannot change the device type to T1 3CH - it stays at sonoff basic no matter what I feed the web interface. All other parameters are saved permanently. Strange.