arendst/Tasmota #1442: Sonoff Touch will not flash

veames, 2017-12-27:
Hi all, I am using the Adafruit Serial Ttl to flash a Sonoff Touch USA. I’ve hooked up the pins as suggested with TX to TX and RX to RX. When it boots normally it flashes once per second when I ground GPIO0 it flashes twice and then once. I assume this is flash mode ? 

In either case I am using atom and PlatformIO. I get an error opening com5 error however com5 is the correct port.  Any ideas out there?

veames, 2017-12-27:
Link to my serial adapter. https://www.adafruit.com/product/954

veames, 2017-12-27:
Also forgot to mention that I know the adapter is 5v. I am using a buck converter to get it to 3.3v

reloxx13, 2017-12-27:
https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation#serial-connection

webtop, 2017-12-28:
You can flash at 5v without any issues, but running a device at 5v causes too much heat, and the device will shutdown for a period, if there is not adequate dissipation of that heat. When I say "device" here, I don't mean the Sonoff, I mean the ESP chip itself. 

But to put it simply...your connections are backwards

davidelang, 2017-12-28:
hooking 5v to the ESP chip can fry it extremely quickly (seconds), so if you try 
to flash using 5v you will probably end up with a dead system.

webtop, 2017-12-28:
I have a whole host of ESP-12 and ESP-7 modules that I flash using 5v, and have not "fried" a single one.
There is a big difference between "operating voltage" and actual allowed voltage. Hooking an ESP8266 up to 5v not fry it, and hooking one up to 6v will cause a "snapback" to ground and reset the chip. On occasion that will also "brick" the chip, but the chip can still be recoverable with a flash erase.

Many chips like the ESP, and AT allow high voltages to "unbrick" a device.

veames, 2017-12-29:
Thanks for the tips guys. Still no luck. I've reversed the TX/RX wires and am still getting

`warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed`

I am wondering what the different interval of flashing lights mean? Normally it's one blink every 2 seconds, sometimes its 2 blinks then 1.

veames, 2017-12-29:
I think I found the problem.
[image]

rene9900, 2017-12-29:
what was the problem? or solution? i have same issue with my sonoff touch.. :(

veames, 2017-12-30:
For me because I am using a different chipset the orientation is different. In my photo I am using the wrong gpio. Actually the GND is on the adjacent side.

webtop, 2018-01-11:
That pin you have in the pic is indeed GPIO0. Ground it when booting, then 2/3 seconds later, remove it from ground. I just successfully flashed a 3 gang T1 and it seems to work perfectly,

By the way...where did you buy that from? It looks like a piece of garbage. Way too much solder everywhere

veames, 2018-01-11:
@webtop mine is the clearer photo. I thought I had fixed the issue due to using the wrong pin, but apparently not, I am still unable to get it into flashing mood.

webtop, 2018-01-11:
Let's make sure all these assumptions are correct:
You have a 4 or 5 pin header soldered to your board
You have GND, RX, TX and 3.3v connected to the header via a USB TTL cable to your computer
The RX from the board is connected to the TX on your cable, and the boards TX to cables RX
You have GPIO0 connected to ground
You plug in your USB cable, wait a couple of seconds, and remove the wire connecting GPIO0 to ground

That should put you into flash mode. On mine I see no indication that I am in flash mode at all.
Before flashing make sure that you have the default configuration set, so that your PlatformIO ini file says the board type is esp01_1m

Are those correct?

veames, 2018-01-12:
Will try to test that this weekend. Will get back to you, thanks! I was hopeful that the LED light would give some indication of what mode the ESP was in.

vvallee, 2018-01-14:
Success! Thanks so much for your help. I think the problem was a combination of not having the RX/TX wired correctly and unreliable connections through my breadboard (I am using a downconverter to get the voltage to 3.3v) At any rate soldering them correctly and shorting the right GPIO for 3 seconds finally did the trick.

The main issue was that my sonoff looks different from all the pictures I have seen. I had thought GPIO0 was on the left where if you look closely there is a black mark. When actually the correct one to ground was on the bottom.

See attached.
[image]

vvallee, 2018-01-15:
Well, now that I got that figured out, I cannot get the sonoff to power on via mains. 
I've wired the 2 live wires (Black) to Lin and Lout and the neutral (white) to Nin.
No power or lights on the sonoff.

vvallee, 2018-01-17:
Fixed it. I was wiring correctly. I just didnt realize it had power as none of the lights flashed initially. Once I hit the touchpad it fired up perfectly.

closed 2018-05-17 by vvallee