arendst/Tasmota #1446: [feature] Watchdog for connected device

glogiotatidis, 2017-12-27:
Introduce a watchdog for the connected device which will trigger a hard reboot if an event hasn't been received for a predefined amount of time.

Example:
 - A RaspberryPI with KODI is powered by a Sonoff S20
 - Every minute using a cronjob the RPi pings a URL (or sends a MQTT packet) on the Sonoff that resets its timer.
 - If for whatever reason the RPi fails to ping the URL, Sonoff powercycles and hard resets the RPi.
 - Sonoff should check if it's connected to the network (and/or can reach MQTT server) before resetting to avoid continuous resets in the case where the whole network (or the MQTT server) is down

I find this functionality extremely useful to re-calibrate systems that have reached an unrecoverable state.

Frogmore42, 2017-12-27:
Since you don't want to do anything if MQTT is down, just run node-red on that server and you can get much better control that way.

Frogmore42, 2017-12-27:
My point was, run node-red on the MQTT server and have the device you want to watchdog send something to that server/device.  You can have great control that way.  I know it will not solve some problems, but this solution is available today.

I know watchdogs can be very useful.  Someone else raised an issue about one he had created, but it had very different requirements from what you said you wanted the first time.  Making one that is both generic enough and simple enough to be useful to many is not easy.

Finally, a poorly implemented watchdog is often worse than none at all, as it will give you a false sense of security and/or reboot when it was not necessary to do so, which can cause other issues.