arendst/Tasmota #1447: Magic Home Module isn't found by Alexa

MischaTyde, 2017-12-27:
I flashed my Magic Home Controller with the latest version 5.10d and set emulation to Hue Bridge.
I'm not able to find the device in my Alexa app. If I configure the device to Sonoff Basic it will be found as a Switch. (If the switch is off the LEDs are red, if the switch is off the lights are green). Same for Emulation Belkin WeMo.

Is there actual no support for using Hue Bridge with Magic Home

AzonInc, 2017-12-27:
@MischaTyde How did you get that actually working? Have you changed the wifi settings before flashing?

MischaTyde, 2017-12-27:
@AzonInc its not really working ;-) 
I set my wifi settings in user_config.h before flashing.

AzonInc, 2017-12-27:
Thanks man - I got it setup for now - now theres only a problem with the gpio - magic home has only the option for 2 gpios

AzonInc, 2017-12-27:
have you got the basic configuration for a magic home led controller?

MischaTyde, 2017-12-27:
My one seems to be from Zengge:
[image]

MischaTyde, 2017-12-30:
After flashing the latest version 5.10.0g my echo was able to find the controller, but it still not works.
Alexa shows the name "LED Controller 1CFA7" instead off my frindly name. And I'm still not able to switch the light on.
I faced an other issue. If I try to set some commands in the console from the Command table or via MQTT (e.g. Color, CT, Dimmer or Fade) I always get the error Unknown Command

AzonInc, 2018-01-01:
try "setoption15 on" - this will enable the commands for you

closed 2018-05-15 by stale[bot]