arendst/Tasmota #1450: Belkin Wemo Emulator ohne Funktion

Egal77, 2017-12-27:
Hallo,

wenn ich Belkin Wemo Emulator einstelle, hatte ich gehofft, dass der Sonoff auch sich als Belkin zu erkennen gibt. 
Unter Alexa wird er gefunden, doch habe ich ein SmartHome  ... nennt sich homee... das eigentlich einen Belkin Steckdose erkennen sollte. Dies ist aber nicht der Fall.

Wenn ich ein NodeMCU mit einer Belkin Emulation aufspiele (kein Tasmota) dann findet mein Server diese. 

Kann es sein das die Emulation einer Belkin Steckdose nicht 100% umgesetzt ist? 
Danke für die Hilfe im voraus

webtop, 2017-12-28:
Ich spreche kein Deutsch, das ist eine Übersetzung, also hoffe ich, dass es Sinn macht.

Sie müssen sowohl die Broadcast Nachricht vom Gerät als auch die Initialisierungs Nachricht von beiden Geräten protokollieren, damit Sie sehen können, ob die Ergebnisse identisch sind. Es könnte ein sehr kleiner Unterschied, der macht ein Gerät n ot erscheinen zu Alexa.

Syberteck, 2017-12-29:
https://github.com/Syberteck/esp8266-Smart-Plug

closed 2018-05-12 by stale[bot]