arendst/Tasmota #1451: Using Sonoff Basic trying to connect GPIO14 to capacitive touch module.

Nargousias, 2017-12-28:
Using a Sonoff Basic and connecting a TTP223 Capacitive Touch Module to GPIO14.  Works if I connect the output in parallel to the GPOI0.  My relay toggles each time I touch the module.  Cant get it to work on GPIO14.   Have tried settings of Button2 thru Button4.  Does not toggle.   Even tried Switch1 it does toggle the relay but only to connect and then immediately disconnect.
I am sure that the TTP223 is going high at touch time. I can wire it in parallel to GPIO0, but would prefer to use GPIO14. 
I have tried switchmode Toggle, Pushbutton and PUSHBUTTON_TOGGLE at upload and all settings on the internal web configuration settings area.   Anything I am missing?
I just want GPIO14 to act like GPIO0.

davidelang, 2017-12-28:
set GPIO14 to be switch1 with switchmode toggle.

check that the touch module is not sending multiple pulses.

button2/3/4 and switch2/3/4 would be for relays2/3/4 which is why they didn't do 
anything.

by default, tasmota is looking for the line to go low, so if it's always pulled 
low and only goes high for a short time as you press it, it may not have it 
going high long enough for the system to notice before it drops back low do you 
have any way for it to pull the GPIO low instead of high when it activates?

Nargousias, 2018-01-04:
I was able to set the module to a toggle mode and pulling low by enabling both jumpers A and B on the TTP223 Capacitive Touch Module.

closed 2018-01-04 by Nargousias