arendst/Tasmota #1452: DS18x20 only sensed at teleperiod-1

avrhack, 2017-12-28:
I notice that the current design means new temps are only pulled from the DS18x20 at teleperiod-1, so that when teleperiod expires the data is 'fresh' and sent as the MQTT JSON packet.

However if you send a STATUS10 (or 8) as the code creates the sensor output immediately, it is 'stale data'.

Similarly if you never set a teleperiod, you will never be able to use temperature sensors by requesting ad-hoc data.

One simple fix is that when a 'status 8' or 'status 10' command is sent, set the teleperiod to teleperiod setting-1 and then it will deliver the sensor output correctly, however that would appear as a 'SENSOR' message rather than as a 'STATUS8' or 'STATUS10'.

I'm happy to put some work into this, but as it's part of the core sonoff.ino I don't want to go to that effort unless it's done in a way that you're happy with. So some guidance on how you want me to do it would be appreciated, or if you want to do it yourself that's OK too :)

arendst, 2017-12-29:
As it takes up to 750mS for a DS18B20 to convert the current temperature it makes no sense to wait that long for a current value. That's why  conversion is started 1 second before teleperiod to have the data ready when needed.

With a teleperiod of 10 a status 8/10 report will be ok.

As I do not believe that the sensor is that quick in reacting to an external temperature change a teleperiod of 10 seconds should be just fine to report the most "real" temperature.

avrhack, 2017-12-29:
Agree the temp doesn't change that often, just that if the design is to allow the status to be explicitly queried it's logical that the number returned would have been just captured. And in the case where teleperiod is zero ie no regular updates, that means there is no way to get the sensor data for the 18B20 as it is only that teleperiod that causes an update.

However it's obviously your code and your design so if you disagree with my proposal I absolutely understand - let's just close this one to avoid clutter, although I do think you're wrong on this one :)

closed 2017-12-29 by avrhack