arendst/Tasmota #1453: Change connectivity settings without instant reboot

terba, 2017-12-28:
Hello,

I would like to change SSID and Password on all my installed sonoffs which have only MQTT support compiled in (no webserver). My idea was to issue `SaveData 0` on them to prevent instant reboot when changing SSID or Password (or MQTT settings). But doesn't worked, it rebooted instantly at the first command, so I was not able to change the password to pass the new SSID.

It would be great if I could change connectivity settings without connecting them one by one over serial. Any tips for this?

Thanks is advance!

P.S. I will now try to have the same wifi password on the two APs, change the SSID, reboot, change the password on AP2 to a new one, change the passwords on the sonoffs. But it is a bit complicated and even gets worse if you want to change MQTT settings also.

arendst, 2017-12-28:
Look for the Backlog command which is very handy in this case...

terba, 2017-12-28:
That's what i'm looking for! Thanks!

avrhack, 2018-01-01:
@terba will you mark this as closed please if you've found your solution - helps keep focus on the real issues that are still needing attention......

closed 2018-01-02 by terba

terba, 2018-01-02:
I wrote a short section about this [in the wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features#send-multiple-mqtt-commands-at-once)