arendst/Tasmota #1455: Full Topic template ignored when MQTT stat is published

hyteoo, 2017-12-28:
When a cmnd message is received via MQTT, the message published with the power status is always formed with the old template: %prefix%/%topic%/

Changing the Full Topic to anything else, that gets ignored and the published message always looks like %prefix%/%topic%/ (ex. cmnd/Sonoff/power )

Toggling the power from the hardware button or the button from the web interface results in a correct message - that is formatted according to the Full Topic template.

This was tested with the latest files as of 28-12-2017, built with VS Code/PIO, on a Sonoff basic and a S20.

arendst, 2017-12-28:
Search these issues. You just entered the third issue about the same problem.

closed 2017-12-28 by hyteoo

lichtamberg, 2019-01-21:
How did you solve this? I have the same problem with the latest version and can't find any reason why it's always switched - even when I change it in the web interface.

hyteoo, 2019-01-22:
Hi
I switched over to Espurna a long time ago.