arendst/Tasmota #1456: Feature request - integration with Telegram bots

ilya-fedin, 2017-12-28:
Hello! I think, would be the fine idea to add in a firmware a feature for device management of Sonoff through Telegram bots. It will give the chance to control from any point of the world to people with a private IP address, without rent of VPN or use of a cloud (eWeLink, etc.).

gajotnt, 2017-12-28:
Use a program like Domoticz or HomeAssistant...

cobra85ec, 2017-12-28:
You can also use node-red .. its great

ilya-fedin, 2017-12-28:
No, I ask to make it without additional devices, such as Raspberry Pi or 24/7 running PC.

cobra85ec, 2017-12-28:
Do u have a alexa.

ilya-fedin, 2017-12-28:
@cobra85ec No, of course

ilya-fedin, 2018-05-04:
Any progress?

ascillato, 2018-06-13:
Do you have any explanation, code for esp8266, any example, link?

ilya-fedin, 2018-06-19:
https://github.com/Gianbacchio/ESP8266-TelegramBot
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot
https://create.arduino.cc/projecthub/JohnWick/telegram-bot-with-esp8266-dbada8
And more on Google

ascillato, 2018-06-19:
Ok,

So, you want to use the instant messaging application for IOS and Android called Telegram with a chat bot inside Tasmota, right?

What do you want the bot to do? write directly to the Tasmota console?

rustymh, 2018-07-18:
May I add to this topic. Telegram will be a great add-on. I'm hoping to add option myself. 

The idea will be to have a separate section for bot token and Webhook link for user input via Web-manager. 
In the code similar MQTT functions can be implemented for Telegram, commands like /on, /off and /toggle. 

This will enable the option to send basic function commands via telegram to the sonoff for switching when not close to the divice.

ascillato2, 2018-08-29:
Hi,

This is a very nice idea but it is out of the actual scope of Tasmota (from [readme](https://github.com/arendst/Sonoff-Tasmota/blob/development/README.md)):

> Alternative firmware for ESP8266 based devices like iTead Sonoff with web, timers, 'Over The Air' (OTA) firmware updates and sensors support, allowing control under Serial, HTTP, MQTT and KNX, so as **to be used on Smart Home Systems.**

So, the label _requested feature (hold over)_ has been added in order to be taken into account.

Thanks for sharing your ideas.

closed 2018-08-29 by ascillato2