arendst/Tasmota #1457: Tasmota T1 3 gang not reporting POWER on gang 1

RichardShipman, 2017-12-28:
Hi, love the work here.
Currently using 5.9.1f on a number of devices and have run into a small issue.
On a Sonoff T1 3 gang, the first gang is not reporting the POWER result via mqtt when the switch is changed, but 2nd and 3rd gang are working fine.  This happens on both of my T1 3G switches.
I've tried paying with SwitchMode and SwitchTopic but to no avail.
I jiust want to be able to recieve the POWER result so that it can switch a remote Sonoff Basic module, I've worked around it by using gang 2 for now, but it basically leaves the first gang as un-usable.
Is there something that I'm doing wrong, or is there an issue with this module?
Thanks

webtop, 2017-12-28:
So the switch is working just fine, from a manual perspective, but because you don't know it's state, you cannot control it via MQTT?
Can you publish to a topic that controls the switching? Does that work?
I'm just trying to understand if this is a "reporting" issue, or a connectivity issue.

RichardShipman, 2017-12-30:
I want to use the switch to control a remote sonoff basic module.  I don't have anything connected to the relay on the switch itself.  I want the switch to report it's status to openHab which will then switch the sonoff basic module on the lamp fitting.

My T1 3 gang is set up as Switch6, 7 and 8 as the 3 gangs.
When I turn on Switch 7 (the second gang of the switch) I get 2 messages from the switch, one from the verbose listener on the switch which reports the channel and an ON/OFF event from the switch element.
2017-12-30 14:47:34.814 [ItemStateChangedEvent     ] - Switch6_Verbose changed from {"POWER2":"ON"} to {"POWER2":"OFF"}
2017-12-30 14:47:34.827 [ItemStateChangedEvent     ] - Switch7 changed from ON to OFF

When I change the status of the first gang for the switch, I only ever get the first message from the verbose listener, and not from the switching element, so the rule listening for the state change never fires.
2017-12-30 14:47:38.646 [ItemStateChangedEvent     ] - Switch6_Verbose changed from {"POWER1":"ON"} to {"POWER1":"OFF"}

All the gangs are set up the same way, and I have another 3 gang which does local switching on gang 1, so it doesn't bother me too much that the status isn't reported correctly, but would be nice if it was.

I can control the switch remotely from OpenHab no problem using MQTT, just don't get the messages back to trigger the rules.

Hopefully this helps you narrow down the cause.  Thanks.

RichardShipman, 2017-12-30:
My mistake,  I have done more investigating and found that my configuration in openhab was at fault, I was looking for POWER rather than POWER1 - D'OH.
Apologies, thanks for bearing with me - MQTT and OpenHAB is a bit of a handful to get to grips with, even as a software engineer of 30 years experience.

closed 2017-12-30 by RichardShipman

gajotnt, 2017-12-30:
Try domoticz :P