arendst/Tasmota #1458: Loses Wifi after 8 minutes

madtech360, 2017-12-28:
I have 5 sonoff basics installed and working with Home Assistant. I am having a wierd issue where all the 5 devices loses wifi after being connected for 8 mintues. Exactly after 52 minutes, they connect back to wifi. This repeats the whole day.

I have device tracker enabled in HA setup and i can see this pattern for all the 5 devices in the history. Any idea what cluld be causing this? 

These devices are configured with static ip and connect to a AP in the upper level of my house. The AP is hardwired to my router in the lower level. The router acts as DHCP server and the static IP reservations are made in the router. I had the DHCP lease time set at 10 minutes and have since changed to 1 day to see if that has any impact. 

I will try to add a screensot from my HA to show the connect and disconnect pattern.

Frogmore42, 2017-12-28:
Since all 5 are doing it with a very consistent pattern, it doesn't sound like a Wi-Fi interference issue.  Have you verified they really are losing Wi-Fi connection?  Or, is it just MQTT?  Have you looked at the MQTT logs?  Have you looked at the Sonoff logs?  Have you tried pinging the Sonoffs when they are "offline"?

webtop, 2017-12-28:
If you have MQTT set up, listen to "tele/[your device topic]/RESULT", and pull from that the RSSI value. Log that for a while and see if there is a dropout on that.

madtech360, 2017-12-28:
I verified a few things. I think my router (Xiaomi Router 3) is at issue here.  Even when the router reports my device to be offline, i am able to open the web interface of sonoff and control the sonoff through my HA interface. 

Not sure why my router behaves this way. 

The reason why i thought i was losing wifi was because i have frequent issues in reaching the web interface. Its like the web server is down. I need to verify when that happens, if i am able to control thru mqtt.

The other thing i noticed in the sonoff web interface console log is that it loses the MQTT connection and it takes 7 or 8 attempts to connect. It has a return code (rc) 2.

madtech360, 2017-12-28:
Here is the sonoff log regardging mqtt connect error:

23:23:10 MQT: Attempting connection...
23:23:16 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:23:27 MQT: Attempting connection...
23:23:33 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:23:44 MQT: Attempting connection...
23:23:51 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:24:02 MQT: Attempting connection...
23:24:08 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:24:19 MQT: Attempting connection...
23:24:25 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:24:36 MQT: Attempting connection...
23:24:42 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:24:53 MQT: Attempting connection...
23:25:00 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:25:11 MQT: Attempting connection...
23:25:17 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:25:28 MQT: Attempting connection...
23:25:34 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:25:45 MQT: Attempting connection...
23:25:51 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec
23:26:05 MQT: Attempting connection...
23:26:11 MQT: Connect failed to 192.168.1.251:1883, rc -2. Retry in 10 sec

I am running a embedded MQTT in HA on a old mac mini

webtop, 2017-12-28:
To troubleshoot MQTT connection issues/dropouts, I would recommend this resource: https://www.ibm.com/support/knowledgecenter/SSFKSJ_8.0.0/com.ibm.mq.tro.doc/q039400_.htm

webtop, 2017-12-28:
For the return code https://github.com/mqtt/mqtt.github.io/wiki/extended_connack_codes
I would log into logging with as much verbosity as you can get, just on a single client, to see if that hints at a reason

madtech360, 2017-12-29:
I can confirm that the device does go offline and my router seems to report it correctly. It is a different pattern today, but consistent across all devices.

https://drive.google.com/file/d/1KY1s4_2-lbtgAxjN7hm_IvMQBaS6dFX_/view?usp=drivesdk

https://drive.google.com/file/d/1k1Q1xm7OhGn5LSv0bSgJBDGaPIImwuV_/view?usp=drivesdk

https://drive.google.com/file/d/1GpuLrnxqdTMv621O8Uy-jZGgS0HIZIZO/view?usp=drivesdk

Any idea?

I cant ping, no mqtt connection

madtech360, 2018-01-01:
After swapping my AP in upper level with a different router(configured as AP) i had lying around, the sonoff have not gone offline. Seems the AP was the culprit dropping the sonoff. Strangely enough i had no issues with other devices(like apple tv) connecting through this AP.

With the new AP, the sonoffs are stable ans running wihout disconnects for 24 hours.

Thanks for the help folks. I now have three MQTT based sensors setup in my HA to monitor the sonoff with Tele State and LWT messages.

closed 2018-01-01 by madtech360