arendst/Tasmota #1459: Sonoff DUAL sometimes doesn't switch

mogwai2, 2017-12-28:
Hi. Occasionally my sonoff dual (non R2) doesn't switch. Today the GUI on one of my sonoff's showed that channel 1 was ON while the connected lights were still off. Toggling channel 1 off and back on using its GUI did switch them on. So it seems that there is an occosional communication issue between the ESP and microcontroller that is controlling the relays...

I already made sure that seriallog was set to 0. I'm running firmware version 5.9.1.

Any idea's?

webtop, 2017-12-28:
Do you have MQTT enabled? If so, have you tried subscribing to see if there is a reproducible issue? Are you logging for a period?
Reproducing the issue is the first step in resolving the issue.

mogwai2, 2017-12-28:
Hi. Yes I have MQQT enabled. I control the Sonoff DUAL using Domoticz / mosquitto. But is issue is not with MQQT. If you send the "ON" command via MQQT then the Sonoff/Tasmota webGUI shows "ON" and if you send the "OFF" command then the webGUI shows "OFF". But the actual relay on the Sonoff DUAL doesn't always switch....

webtop, 2017-12-28:
Have you tried playing with the QoS level on the client? It may be missing a message and just ignoring that fact if the QoS is zero

arendst, 2017-12-29:
It's probably a serial communication problem between the esp8266 and the microcontroller chip within the Dual.

Solved in the Dual R2.

mogwai2, 2017-12-29:
Hi Theo, many thanks for your reply. Unfortunately I don't have the R2 and have multiple DUALs built into my house. Is there any way of reading back the status of the microcontroller so you can check if the communication was OK and re-send of not OK? Or can you bluntly send the switch commands from the ESP to the microcontroller multiple times to become more robust against serial communication errors?

I'm very happy with my Sonoff DUALs + Tasmota, but the occasional hickups are quite annoying...

Kind regards, Bastiaan

mogwai2, 2018-01-30:
It seems that the problem is caused by the Energy Saving that I was using. Setting `sleep 0` to seems to have solved the issue.

closed 2018-06-14 by stale[bot]