arendst/Tasmota #1460: Add IR Support for iLife / Medion vacuum robot

BernRos, 2017-12-28:
Someone programmed an IR control for iLife / Medion vacuum robots with an ESP8266.
(http://www.roboter-forum.com/showthread.php?16821-Erfahrungen-mit-chuwi-iLife-x5-(medion-Hersteller)-vorhanden&p=220675&viewfull=1#post220675)
Source Code is available in post.
So I flashed the software to a Sonoff Basic, which worked like a charm. Unfortunately my programming skills are very basic. Could someone assist how to integrate this piece of Software in the Sonoff Tasmota master?
My plan is to integrate a Sonoff SV in the loading station of the robot and to add IR Diodes in some Sonoff S20 to control the robot when he is on it's cleaning cycle.

webtop, 2017-12-28:
I'm not sure if you'll get that into master, as it seems like an edge use-case, rather than a general use-case. 

What you need is a module to integrate with Tasmota, but can be kept in a 3rd party repository, and either subscribe to events, or use system hook. I'm not sure if Tasmota broadcasts events, or has a hooking system in place, as I am not fully familiar yet with its architecture.

BernRos, 2017-12-28:
I have seen this IR HVAC option,  which also looks like a edge use-case. So I thought that maybe more users are interested in a option to control vacuum robots. 
Unfortunately I don't have any idea how to start to build my own version with this IR remote included.

closed 2018-05-11 by stale[bot]