arendst/Tasmota #1461: Sonoff TH16 - WiFi not working

PlumbumDN, 2017-12-28:
Hello there,

for x-mas a brand new sonoff TH16 has arrived on my desk.
I was curries what to do with it, but failed at the initial stage. The Sonoff seems to dislike my wifi, failing for the main AP (time out) and another AP (sonoffIoT) I set up to test some things out, telling me, that the password is incorrect. I triple checked the password on the AP and sonoff and reflashed and cleared flash a dozen times. And now I don´t have a clue, why the connection is failing over and over again.

What have I done before?

1.  Set up new AP with different configs and non special character password
WPA, WPA2; AES+TKIP, only TKIP, only AES

2. reflashed different Versions of Tasmota 5.10

3. cleared config and flash between uploads

4. checked soldering points on the TH16

The TH16 goes to WPSConfig after failing to connect, and on the web panel I see all Wifi APs next to the device, including my own ones...

What could I try next?

Best regards,
Robert

```
--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00:00:15 WIF: Verbindung fehlgeschlagen da der AP nicht antwortet (timeout)
00:00:15 WIF: verbinden mit AP2 sonoffIoT in Modus 11N wie sonoff-3273...
00:00:24 WIF: Verbindung fehlgeschlagen mit AP weil das Passwort falsch ist
00:00:24 WIF: WPSConfigaktiv für 3 Minuten
00:00:43 APP: starte neu

 ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

PlumbumDN, 2018-01-04:
Ok, my last guess, that DHCP-Server could be to slow for the sonoff to fully join the network didn´t change anything. Also with a fixed IP I´m not able to get this thing online...

Any help or idea is welcome.

Best regards,
Robert

`Windows PowerShell
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

PS C:\Users\Robert Plum\Downloads\Sonoff-Tasmota-development\Sonoff-Tasmota-development> pio device monitor --port COM5 --baud 115200
--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00:00:15 WIF: Connect failed with AP timeout
00:00:15 WIF: Connecting to AP2 sonoffIoT in mode 11N as sonoff-3273...
00:00:30 WIF: Connect failed with AP timeout
00:00:30 WIF: WPSConfigactive for 3 minutes
00:00:49 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16

tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_8B2CC9, GroupTopic sonoffs) Version 5.10.0h
00:00:00 WIF: Connecting to AP2 sonoffIoT in mode 11N as sonoff-3273...
00:00:15 WIF: Connect failed with AP timeout
00:00:15 WIF: Connecting to AP1 4CE676648F58 in mode 11N as sonoff-3273...
00:00:30 WIF: Connect failed with AP timeout
00:00:30 WIF: WPSConfigfailed to start
00:00:35 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)`

Frogmore42, 2018-01-06:
What channel is your AP on?  What other networks are present?

The log indicates it is timing out connecting.  Have you tried setting up a test network with no security to see if that is the problem?

Have you looked at the log in your AP/router to see what it thinks about the connection attempt?

I have found the esp8266 to connect pretty reliably, but I have also had issues when there was not enough power and also when I had a bad esp8266.

If you only have one device, it is really hard to figure out the cause of the issue.

PlumbumDN, 2018-01-06:
Hi @Frogmore42 
I could have come up with it myself and tried to use unencrypted WiFi.....
If i disable WPA2 it directly connects to the networking resolving an IP address over DHCP within seconds.

The APs I tested are on channels 1, 3, 9 and 11
There are numerous networks present here, most on channel 9

The router didn´t log anything. I can´t see any try to connect... when security is enabled.

Unfortunately, at the moment I only have one device to test tasmota on, but I will get another one to check if the issue persists. But issue #1366 looks quite similar...

PlumbumDN, 2018-05-07:
As an update: A Second and third Sonoff work just fine...
Only the First One only works in unencrypted WiFi.. .

ascillato, 2018-06-13:
Hi

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-27 by ascillato2