arendst/Tasmota #1462: Witty Cloud + WS2812 - not working / configuration issue?

alexjoni, 2017-12-29:
Not sure this is a bug, or a misconfiguration on my part,
but I wasn't able to drive a WS2812 light strip on a witty cloud (module: 32).

I connected the WS2812 strip to GPIO03 (according to comments in user_config.h - along with USE_WS2812_DMA).
After enabling pin as WS2812, I see a slider and a TOGGLE button - but it either drives the onboard LED (GPIO02 - D4 Blue led) or the onboard PWM LED (GPIO12,13,15).

I was able to drive the WS2812 after changing the definition in sonoff_template.h:
  { "Witty Cloud",     // Witty Cloud Dev Board (ESP8266) - https://www.aliexpress.com/item/ESP8266-serial-WIFI-Witty-cloud-Development-Board-ESP-12F-module-MINI-nodemcu/32643464555.html
     GPIO_USER,        // GPIO00 D3 flash push button on interface board
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        //AJ: GPIO_LED1_INV,    // GPIO02 D4 Blue Led (0 = On, 1 = Off) on ESP-12F
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_KEY1,        // GPIO04 D2 push button on ESP-12F board
     GPIO_USER,        // GPIO05 D1 optional sensor
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_USER,        //AJ:      GPIO_PWM2,        // GPIO12 D6 RGB LED Green
     GPIO_USER,        //AJ:      GPIO_PWM3,        // GPIO13 D7 RGB LED Blue
     GPIO_USER,        // GPIO14 D5 optional sensor
     GPIO_USER,        //AJ:      GPIO_PWM1,        // GPIO15 D8 RGB LED Red
     GPIO_USER,        // GPIO16 D0 optional sensor
     GPIO_ADC0        // ADC0 A0 Light sensor / Requires USE_ADC_VCC in user_config.h to be disabled
  },

arendst, 2017-12-29:
I do not support both PWM and WS2812 at the same time as they both use the same light handler. As the default witty module is configured with PWM it means you cannot attach a WS2812 string to it.

Looking at your suggested changes it seems to me it's become a wemos d1 mini module and makes the witty module obsolete. Fine with me as it saves another 32 precious bytes if I remove the witty module.

alexjoni, 2017-12-30:
I have no preference either way. I think for beginners it's better to see witty cloud in the list of supported devices, but there should be some information somewhere that you can't use WS2812 if your module already has some PWM light defined...

closed 2017-12-30 by alexjoni