arendst/Tasmota #1464: Animations on WS2812 freeze periodically together with #define USE_WEBSERVER

alexjoni, 2017-12-29:
Having #define USE_WEBSERVER active together with WS2812 animations (Scheme > 6) leads to periodic "hiccups" in the animations.
Especially on 8,9,10,11 (animations with scrolling), if you have a faster speed (speed 1..5) you can see the animation is smooth for 2-3 seconds, then it halts temporarily, then continues.
Disabling the Webserver in the code leads to smooth animations, and all seems ok.

I will try to pinpoint what hangs in PollDnsWebserver();
But maybe someone has an idea!?

arendst, 2017-12-29:
Close your webpage during animations or do not select root/console page as they draw data every 2.34 second from the webserver probably leading to your experience.

Best to not use the webpages at all and enjoy MQTT as that's tasmota core business.

alexjoni, 2017-12-29:
Thanks for looking at this.
What I reported (animations having hiccups) was with the webserver enabled, but no active webpage opened, and surely not with the console.
Initially I had the webpage open, and the hiccups kinda coincided with the ADC measuring (VCC and later light sensor on the witty), but I used commented out //#define USE_ADC_VCC and I removed the ADC pin from the witty definition - this should have disabled the ADC, right?
MQTT stuff doesn't affect animations in any way.. that's great, and what I care about.

I will try though to see if I can pinpoint where the slowdown happens.

alexjoni, 2017-12-30:
Commenting out #define USE_EMULATION stopped the slowdown.
After that I re-enabled it, in the end it worked with HUE emulation (not really sure why) - but with all browser windows closed. 
If there's a browser window (or an alexa command) it stops a bit, but not so that it's a problem.

closed 2017-12-30 by alexjoni