arendst/Tasmota #1466: Unable to use apostrophes in friendly name

idkpmiller, 2017-12-29:
If you use a name which includes an apostrophe it will take the name but then when you return to the Configuration Other it shows the name up to the point oif the apostophe onle.

Example:

Set Friendly name: Holly's Fan
Name when you go back to check is set as: Holly

This appears to affect all unit types.

Thanks

Paul

arendst, 2017-12-29:
As designed as that's one of the pretty programming special characters which I do not escape.

You may want to try a name like ``Hollys Fan`` or ``Fan of Holly`` ;-)

idkpmiller, 2018-01-12:
I had to change my naming scheme for my entire house across multiple automation vendors because the devices with tasmota firmware did not escape the name correctly. I feel this is an issue or you may call it an enhancement either way, it would be better changed than left as it is for others.

Regards

Paul

closed 2018-05-26 by stale[bot]