arendst/Tasmota #1467: TH module not picking up sensor

idkpmiller, 2017-12-29:
I ghave 6 other TH10 modules all are working fine except one.
All TH modules worked prior to changing from ESP Easy firmware to tasmota 5.10c
The sensor data does not show on the web page for the affected device.
I have reflashed the device multiple times and it seems to be the same.
I have also tried a different sensor with no luck.

Regards

Paul

Frogmore42, 2017-12-29:
Since only one is not working, it points to an issue with that one.  Since you have reflashed it multiple times, it is looking like a hardware issue.  I had one TH module that has a bad 2.5mm socket.  It would work sometimes, but not reliably.  I added a 3.5mm one and have had no more issues

idkpmiller, 2018-01-12:
I too felt that this was most probably a hardware issue but thought it worth bringing to your attention in case others had seen the issue.

I have now confirmed this is a software bug not a hardware  issue.
I purchased another TH10 and then set it up with the same sensor and the same thing happened the sensor was not seen. I then tried another sensor from a working TH10 and it also did not work this is where I previously opened a ticket as I thought it may have something to do with the firmware upgrade I had previously done.

After some investigation it turns out the bug is repeatable for me using a TH10, tasmota 5.10c and a DHT22 sensor.

1. set the TH10 back to a basic module
2. Enter the configuration menu>Configure Module and set the module to TH, AND also set the GPIO14 sensor parameter to 03 SI7021 
3. select save

now when the module comes back the sensor will not initialise and display its readings.
to make the sensor work step 2 must be performed in two steps not one.
set the module type then save, then once it returns set the sensor type and save again, then the sensor works fine.

Regards

Paul

Frogmore42, 2018-01-14:
Pretty sure that is expected behavior.  The module type takes priority and the configured pin(s) need to be set after you set the module type.  I am pretty sure the pins get set to defaults for the new module type and that is visible in the UI, but perhaps not obvious.

idkpmiller, 2018-01-15:
That is not apparent from the presentation of all the parameters at once. I
did not find this part of the design specified in the WIKI either.
This is an area for improvement, personally I call it a bug, but you make
that call not I

My job of notifying you is complete, I will not fall into the same trap
again.

Regards

Paul

closed 2018-05-29 by stale[bot]