arendst/Tasmota #1469: Two Sonoffduals communicating

sammyke007, 2017-12-29:
Hi everyone.

I've ordered two Sonoff Duals (R2) to control my lights.
I've read all of the wiki a dozen of times and hope that this configuration is correct? 
The wallswitch currently available at Sonoffliv only has 1 hardware switchwire to "light living".
The wallswitch currently available at Sonoffeet only has 1 hardware switchwire to "light eetkamer".

I want to control both lights from both wall switches (and from Domoticz afterwards) without hassling about wether the hardware switches are on or off. I know that i must use TOGGLE for that. 

[image]

I hope my sketch is clear?

Tnx in advance!

sammyke007, 2017-12-29:
Damn... I seem to understand that I cannot use switch 1 local for relay 1 (Switchtopic = 0) and switch 2 to control another Sonoff Dual's relay 2 (Switchtopic = Sonoffeet)? 

Any way to be able to do his?

gajotnt, 2017-12-29:
Why not use a software like Domoticz, and set the 2 duals to use the same IDX and there you go problem solved

sammyke007, 2017-12-29:
It is intended to be used on my Domoticz. Can you get a little bit more into detail please? Tnx for taking your time!

gajotnt, 2017-12-29:
I think it should work well this way. 
Where IDX: 22 and IDX:23 is your corresponding devices in domoticz

[image]

sammyke007, 2017-12-29:
That seems to be a great solution! Can't wait for the (hopefully R2) duals to arrive!
I guess SwitchTopic doesn't need to be the same for the domoticz config to work?

gajotnt, 2017-12-29:
In this case since for all effects you only need one relay (one for LightLiving and another for Light eetkamer) a sonoff basic should have been enough and smaller. And the switches could be pluged to GPIO 0 and GPIO 14.
But this way you have will also do :)

sammyke007, 2017-12-29:
I tought of that too, but the Basic has 1 relay, and as far as I know, that wil give problems with 2 switches and 1 relay. In the situation above pressing the right switch on either of the wall switches, activates both relay 2's where one of the two is not connected and vice versa. 

When I have 2 switches and 1 relay, switch 1 (GPIO0) will switch relay 1, switch 2 (GPIO14) will switch relay 1 too.
I can't connect the hardware wire on the relay of the second basic because it will always be relay 1... 

I need relay 1 to be switched on 1 wallswitch by switch 1 and relay 1 from the other wallswitch to be switched by switch 2. That is not possible AFAIK.

Or am I wrong?

gajotnt, 2017-12-29:
Try to see if you understand my logic

[image]

gajotnt, 2017-12-29:
It should work like this, since i have in my bedroom a basic, that turns on the room light and turn on another sonoff basic that controls the bedside lamp

sammyke007, 2017-12-29:
I think I completely understand!
I made a little text change in your graph. 

So in 1 Sonoff the Switch1 is IDX22, in the other Sonoff Switch 1 is 23. Thank you gajotnt!!!

[image]

gajotnt, 2017-12-29:
Yeps, thats the idea :)

sammyke007, 2017-12-30:
I've tested it today on a Wemos D1 mini, configured as a Sonoff Basic with 2 switches and it works!
Switchtopic needs to be set to 1 and serial logging needs to be turned off. Then I configured GPIO 14 as Switch1, GPIO1 as Switch2 and GPIO3 as Relay2 (dummy/not connected). On my Wemos I connected a LED to GPIO12 (Relay1) for testing.
In Domoticz I configured both IDX1 and IDX2 and that's all! I now have 2 toggles and I can toggle both Domoticz devices with the switches. Tnx A LOT gajotnt!

closed 2018-02-16 by sammyke007