arendst/Tasmota #1470: "wifi lost / back to AP" major issue

pliviu, 2017-12-29:
If sonoff loses wifi connection to your router (the router resets or crashes or whatever) it become AP (access point) and this is not OK because for a very short period of time the relay change its state (esp reboots) and that affects the device that is plugged in in a sonoff pow.
For example, if the device plugged in in a sonoff pow is a windows server and you reset the wifi, the sonoff become AP (access point), the relay change its state and you'll get blue screen on windows (the windows server crashes ... depending on how premium its power supply is)

Proposed fix:
The sonoff should become AP (access point) if and only if it has no wifi connection AND **its uptime is less then 3-5 minutes**. This way, you can use sonoff for days and if it loses its wifi connection it will not going to become an access point and change the state of its relay.

arendst, 2017-12-29:
See some other issues here with a solution you may have missed. Keyword is command WifiConfig and option 4 or 5.

mariusgeorgescul, 2017-12-29:
I am running version 5.10.0 and I'm at the moment affected by this issue. 
Can you please elaborate on how this can be solved. I assume the firmware needs to be recompiled, but I'm not sure what should be changed in the source code to avoid this.

pliviu, 2017-12-29:
Theo, i cheked the source code and saw that you already implemented WIFI_RETRY, WIFI_WAIT options since version 5.8.0.
Nevertheless, i see in user_config.h that default WIFI_CONFIG_TOOL is WIFI_WPSCONFIG and for that matter my reported issue is still present in 5.10.0.

In my opinion, if sonoff loses wifi connectivity, it should never ever become AP and affect the relay; it should only lose its management and retry to connect to wifi forever until you manage to fix your wifi on the router or you decide to reboot sonoff if you need it configured as AP.

Do you think it may be appropriate to make this WIFI_CONFIG_TOOL configurabile from the sonoff web interface / Configure WiFi in the next release?

P.S. - since i use your software on a few sonoffs and i'm really grateful for that, i want to give you something back ... can you please send me your email address?

arendst, 2017-12-29:
@pliviu WifiConfig is just one of the many commands available online. See the wiki regarding other commands.

The idea is that initially the user uses AP mode (or WPS) and once installed change it online to 4 or 5. All commands can be entered via Mqtt, Serial or the web interface (Console).

You can contact me directly via the link on the Github main page.

pliviu, 2017-12-29:
@arendst I confirm you can change WifiConfig via mqtt and this issue is fixed since 5.8.0.
thanks!

closed 2017-12-29 by pliviu