arendst/Tasmota #1472: Chirp! Plant watering alarm

mredone, 2017-12-29:
Is any chance to add module to Chirp!
It have i2c communication, and it can be connected to wemos. 
Now i have:
`SHT: Sensor did not ACK command`

I found it with i2cscan but it disappear after few second.

[https://wemakethings.net/chirp/](url)

lobocobra, 2018-01-02:
Why don't you buy a Door Sensor from homematic and use that Chirp thing to close the contact on   that homematic sensor? You need to get this device to the wifi anyhow.

As an alternative you could use a sonoff basic as well, with a switch wire to the Cirp. But personally I am not a fan of combinations of 230V and a wather detector. So my first proposal would be safer...
=> There might be also some Sonoff low-voltage solutions you can use for that.

mredone, 2018-01-02:
But it's not a weather but something like mi flora sensor. Tasmota is very flexible firmware. So i can install on wemos, add chirp!, many many other options and send everything to my mqtt broker.