arendst/Tasmota #1473: POLL: While using different locale keep JSON text in English

arendst, 2017-12-30:
From version 5.10.0e I provide the means to keep the JSON text messages in a different language from the locales.

In the locales DE, NL and PL the JSON text is indeed translated but seems to amount to problems with JSON users like node-red not being able to handle German umlauts for instance.

I propose to keep JSON text in English for ALL locales and therefore want to change the DE, NL and PL locales.

So if anyone in favor of English will add his thumb to this issue I will change within the next week.

If there are any doubts please add to this issue and explain. There is always to possibility to provide both languages using a define in the language file but I prefer not to take that road...

gtortone, 2017-12-30:
I definitely agree with separation between web interface labels and JSON attributes name; I'm in doubt on how to handle JSON attributes in messages not related to telemetry like results of MQTT commands...

davidelang, 2017-12-30:
translations are for human interaction, Language choice for the GUI should never change the over-the-wire API

Frogmore42, 2017-12-31:
I agree, programmatic interfaces should be language neutral.  But, UI should be localized to the user's preference.

webtop, 2017-12-31:
Agree. Modern English has no artifacts, and is common choice. I vote for English

avrhack, 2018-01-01:
Agree. JSON in English, UI localised......

lobocobra, 2018-01-02:
I believe also that this is the right way to absract the code level from the UI level.

reloxx13, 2018-01-02:
it also seems, that the iobroker cannot read things like "Current" if the german firmware is used.
Current is Strom in german, so yes, this should be done asap.

closed 2018-01-10 by arendst

reloxx13, 2018-06-26:
old but still: should be the state in the json response (e.g. status 0) changed by statetext if we look at this poll/ticket?

is it possible to send normal on/off per json/status 0 and changed statetexts only to mqtt ?

https://github.com/reloxx13/TasmoAdmin/issues/154#issuecomment-400422874

BR