arendst/Tasmota #1474: Sonoff Basic New Model

catweezle333, 2017-12-30:
Hello,
can someone Help me to identify the RX/TX/GND of the New Sonoff Basic?

[image]
[image]

webtop, 2017-12-30:
That's not an easy task from the images you posted. Can you add a photo of the back of the WiFi module?

alexjoni, 2017-12-30:
I don't think that's an Itead Sonoff.. It's rather a clone..

webtop, 2017-12-31:
Yeah, my thought too. There’s no branding on the board.

Kiwifranky, 2018-01-01:
HI,

I have two modules of the same type. So far I have found only TX. I assume they use RX and D0 for other purposes like LED/Switch/Relay. Have got my lodig analyser ready and will start some testing next week.
So - I am on it and will tell you if I have found out how to program it!

If you should find a solution please inform me too...

So long

Franky

merlin051, 2018-01-02:
This is definitely a clone device.

Note on the plastic case it says Smarthome and not Sonoff.

Kiwifranky, 2018-01-17:
Hi, 
finally I had a little bit of time to play around with the SMART HOME switch which is a clone of the Sonoff Basic, but with another chip.
See pinlayout on the following image...

Hope it helped!

Franky 

[image]

Kiwifranky, 2018-01-17:
Hi again...

To make it work with TASMOTA:

Add  

`  { "Smart Home",      // Smart Home Switch (ESP8285 Sonoff Basic Clone)
      0, 0, 0, 0,
      GPIO_REL1,       // GPIO04 Relais
      0, 0, 0, 0, 0, 0, 0,
      GPIO_KEY1,       // GPIO12 Button on casing
      GPIO_LED2,       // Red Led
      0,
      GPIO_LED1,       // Blue LED
      0, 0
  }`

into the sonoff_template.h ModuleTemplate collection and add  "SMART_HOME" to SupportedModules enum and the nicelist.

Have a nice day....

Franky

Hepman, 2018-01-22:
this does not work for me.

Can someone confirm the pinout ?

Could not flash Tasmota

Kiwifranky, 2018-01-22:
Hi,
I just checked if I used another PlatformIO.Ini, but no... everything as usual.
I flashed mine using this setup but I also had to try 3 times to get it into flashmode by pulling down d0....

Frank

Kiwifranky, 2018-01-23:
Hi again, 
just an idea by looking at the pinout... Maybe I have mixed RX and TX, I will check when I am back home....

Frank

Hepman, 2018-01-23:
I habe tried to Switch rx/tx 

That is not the Problem

Kiwifranky, 2018-01-25:
Hi again,

I reprogrammed mine again and everything worked fine (again).
The layout is OK!
If this is your first device to program at all....

Here is the layout from the bottom side of the pcb: 

[image]

You have to connect:
RX to the TX connector of your serial Interface 
TX to the RX on the serial board
3V3 with the 3.3V Supply
GND with the Ground of your Supply
D0 has to be connected to GND while you power up the device

By the way:  I am using Atom and PlatformIO for programming.

Smoffitz, 2018-01-25:
Great work Kiwifranky.  Can you submit a PR with your changes? This board is becoming more command and it would be nice to have this mainlined so future users could just grab a precompiled binary.

Perhaps this is a dumb question, but how are you connecting your serial device to the board for programming?  I'll probably just solder on some jumper wires and connect to that, but if there's a better way to make temporary connection to the pads, I'd love to hear about it.

Smoffitz, 2018-04-16:
You can use the Generic Module to configure this, no need to patch and compile a special version:

[image]

trigal, 2018-12-01:
i can confirm this trick is working, but the "relay1" for me has 21 as number, not 17. Actually I don't really know what that number means, I followed a semantic schemen and everything was ok. Maybe it is because I did not add the lines inside sonoff_template.h , I leaved everything  as git-cloned :-)

Here's my configuration

[image]

this was my purchase on amazon (italy) JamBer - Smarthome
https://www.amazon.it/gp/product/B07B6GMTR4

> You can use the Generic Module to configure this, no need to patch and compile a special version:
> 
> [image]

genuk, 2019-01-28:
New(?) revision(version) SmartHome available.
Buyied from here: https://www.banggood.com/AC90-250V-WiFi-APP-Relay-Module-DIY-Smart-Home-Automation-Light-Switch-Work-With-Amazon-Alexa-Google-p-1351912.html?cur_warehouse=CN

Pins have to be soldered but they are labeled. To put it in the programming mode you have to shortcut GPIO0 and GND when powering on (3,3V). See photos.
I can confirm this revision also work with the settings from above.

[image]
[image]
[image]
[image]

tbadde, 2019-03-27:
Hello
do you have a reddy projekt (file) that you can send me?
i will flash it with atom.
Thanks Thomas

ttrueten, 2019-07-19:
As genuk said above, it runs well with the [UNEEDE Wifi](https://amzn.to/2Gno9g6) Switch, too, which has the same Layout as the "Smart Breaker".

vzorglub, 2019-08-09:
Template published on the wiki.
'''
{"NAME":"SmartHome Swit","GPIO":[255,255,255,255,21,255,255,255,17,57,255,56,255],"FLAG":15,"BASE":18}
'''
Note that the two leds are inverted