arendst/Tasmota #1475: DHT11 Sensor

profesionalTI, 2017-12-30:
Hi, I have Tasmota 5.10.0f on my Sonoff Basic. I plug a DHT11 Sensor, I config it on GPIO14 but when the Sonoff Restart, I can't see Tº and H on the main home page of my sonoff, some can help me please?

webtop, 2017-12-30:
Have you tried on another GPIO? That would give a good indication where the issue might lie. I have a DHT22 on a GPIO4, and the T & H come up immediately, so it should in your case.

profesionalTI, 2017-12-30:
Thanks, I test, regards.

profesionalTI, 2017-12-30:
The DHT22 si connect to the sonoff basic?

I can’t find DHT22 sensor on GPIO 14 in the new firmware 

Coordialmente,
Best Regards,

Marcelo A. Anuch Medina
Consultor de Empresas Público/Privadas
Ingeniero Comercial, MBA UDD
Business Management

🔒   La información contenida en esta transmisión es confidencial y está dirigida exclusivamente a el o los destinatarios indicados. Cualquier uso por otras personas que no sean sus destinatarios está sancionado por la ley chilena. Si ha recibido esta transmisión por error, por favor destrúyala y notifique al remitente si corresponde. Las opiniones expresadas en este correo son las de su autor y pueden no ser compartidas por la compañía. Marcelo A. Anuch Medina,  no asume obligaciones ni responsabilidades por el contenido de este correo. Tampoco garantiza que la transmisión de este correo sea segura o libre de errores, y en consecuencia, no responde por virus, pérdidas, destrucción, recepción tardía o incompleta.

🍀  Antes de imprimir este correo electrónico asegúrese que sea realmente necesario: Cuidemos el medio ambiente! Todos vivimos en un solo planeta.

arendst, 2017-12-31:
@profesionalTI Try AM2301 option as it uses the same handler as the DHT22

profesionalTI, 2017-12-31:
Thanks Theo, I try, regards

Coordialmente,
Best Regards,

Marcelo A. Anuch Medina
Consultor de Empresas Público/Privadas
Ingeniero Comercial, MBA UDD
Business Management

🔒   La información contenida en esta transmisión es confidencial y está dirigida exclusivamente a el o los destinatarios indicados. Cualquier uso por otras personas que no sean sus destinatarios está sancionado por la ley chilena. Si ha recibido esta transmisión por error, por favor destrúyala y notifique al remitente si corresponde. Las opiniones expresadas en este correo son las de su autor y pueden no ser compartidas por la compañía. Marcelo A. Anuch Medina,  no asume obligaciones ni responsabilidades por el contenido de este correo. Tampoco garantiza que la transmisión de este correo sea segura o libre de errores, y en consecuencia, no responde por virus, pérdidas, destrucción, recepción tardía o incompleta.

🍀  Antes de imprimir este correo electrónico asegúrese que sea realmente necesario: Cuidemos el medio ambiente! Todos vivimos en un solo planeta.

Jabo889, 2018-01-01:
Hi, I have the same issue with DHT11  sensor on Wemos D1 Mini. Tried a few GPIO's with no luck. Therefore relay working fine. Do the Wemos D1 needs to be conected to MQTT?

Any help will be appreciated.

The below is attached screen shot. 

[image]
[image]

phdemartin, 2018-01-05:
Same isue witha sheild DHT11 or DHT22 on my Wemos D1 Mini.

closed 2018-05-19 by stale[bot]