arendst/Tasmota #1476: How to avoid wifi reboot every day, Sonoff-Tasmota's lights are automatically shut down.

1043717432, 2017-12-30:
How to avoid wifi reboot every day, Sonoff-Tasmota's lights are automatically shut down.
How to solve this problem? This problem has a headache to me.

webtop, 2017-12-30:
What's your tech knowledge level? Do you have any investigation notes?
Is the router doing the disconnect, maybe due to a leasing issue? Can you monitor the RSSI value and see if that drops off? Before the disconnect.
Does it happen at the same time every day?

There are a number of things to consider before anyone can help, here.

1043717432, 2017-12-31:
I intentionally turn off the route, disconnect the network with the sonoff, and when the relay is open, the sonoff automatically closes the luminaire without WiFi.

avrhack, 2018-01-01:
Check the PowerOnState variable - it should be the default of '2' in which case the sonoff should remember your power state and not close it down. Sure you know but you can check it by sending topic cmnd/name/PowerOnState where name is the topic name of the sonoff......

1043717432, 2018-01-02:
PowerRetain On
SwitchRetain On
ButtonRetain On
My sonoff mqtt server connection, so every day when the routing restart sonoff switch, it will automatically flash the light bulb, and sometimes the lights will be turned off, the light bulb will be turned off suddenly.

thomasnordquist, 2018-01-10:
I'm experiencing very similar behavior.
It appears that if the esp8266 cannot find any wifi it can connect to successfully, the esp8266 resets.

Scenarios where I experienced resets of the esp8266/relay:

### Scenario 1
2 different Wifi APs configured and expected to have a DHCP. First AP was missing DHCP, the second AP was not present. The ESP resetted itself every 1-3 minutes. I did't time it exactly.

### Scenario 2
I have a raspberry pi in AP mode. While developing I realized when I restart the raspberry pi, some lamps which are connected to a sonoff sockets flicker.

Seeing both scenarios It appears the sonoffs reset when no wifi is present, which is kind of unacceptable from a QOS standpoint.

Currently all my devices run 5.9.0, did't have the time to investigate further.

***Note***: Flickering seems to appear only when the relay is turned on while the AccessPoint is missing. There is no audible indicator of the relay if the socket was turned off.

davidelang, 2018-01-10:
look at wifimode that is what is telling it to do this, if you just change that 
setting, you avoid the reset.

1043717432, 2018-01-10:
@davidelang I don't understand the solution to the problem you are talking about!

thomasnordquist, 2018-01-10:
I'm not entirely sure what he meant with `wifimode`.

He could refer to this: 
https://github.com/arendst/Sonoff-Tasmota/blob/050191671331bd9408489087d0fdcba586dcda7f/sonoff/user_config.h#L57-L58

In my case I should be on the default `WIFI_WPSCONFIG` with all devices, I'll check what happens if I choose `WIFI_RETRY` instead.

arendst, 2018-01-10:
Yes he means the command ``wificonfig``. See the wiki about possibilities but I would suggest you set it to option 5.

closed 2018-05-24 by stale[bot]