arendst/Tasmota #1477: RF Bridge Homebridge Scene

jannnfe, 2017-12-31:
Hello,
I have flashed my Sonoff RF Bridge and learned some old radio sockets. I can switch them all individually, but if I create a scene in Homebridge, which is to turn off all radio sockets, only one of the sockets of the RF Bridge is turned off. Is there any way that you can successfully create a scene that turns all outlets on and off?
Thanks

MatzeJoerling, 2017-12-31:
You cannot start n RF signals at the same time, only one. So you need to delay the off / on commands for at least 0.5 to 0.7 sec per device. I have 1sec delay between the commands, to be save and that setup is working.

regards

jannnfe, 2017-12-31:
Can you tell me how i can set this delay in Tasmota firmware?
What commands i have to use?
Thanks! :)

MatzeJoerling, 2017-12-31:
nope, you have to do that in your home automation.
My node red looks like:
[image]

Dash01 switches all on/off, Dash02 only Video Light.

closed 2018-01-25 by jannnfe