arendst/Tasmota #1478: Sonoff POW: Smoothen values

BlackAlpha700, 2017-12-31:
Hi,

is there a way to get the P, U and I values more smoothen? Even there is no load on the POW (relay switched on), the P value toggles most of the time between 1W and 2W, sometimes it is 0W. But there are some rare peaks, where one single value could be up to 90W - without any load! I think this are measuring errors of the POW due to analogue measuring. This is similar to I and sometimes U.
Actually I use teleperiod 20 to get many values via MQTT into openHAB and there I calculate every minute the average of the last 5min. This fixes the measuring errors slightly with the cost of speed. The values are no more reactive and theres a much higher load to openHAB and the persitence.

I don't know how fast the POW could measure the values, but I think there must be a way to measure some values (e.g. 5 or 10) internally, delete the highest and the lowest value, calculate an average of the reamining values and then send an MQTT telegram.

Is it possible and useful to integrate such a function?

Greetings,
BlackAlpha700

arendst, 2017-12-31:
Has been discussed before. I can't find the issue but try to search on POW.

In short; There is already smoothing present but measuring low power consumptions takes several seconds. That way there is no time smooth out. Pow (=HLW8012) is not made for measuring low power usage.

BlackAlpha700, 2017-12-31:
The HLW8012 is slow, I know. I've read somewhere that it takes 4-5s to measure. Even if someone is using a teleperiod of 30s, there would be 5-6 values for a slight smoothing. What about making this smoothing optional, limiting the teleperiod to a minimum of 30?

lobocobra, 2018-01-02:
IF it is important for you to have a stable output value, then you have the POW anyhow connected to Openhabian or similar, as you are probably not sitting the whole day in front of the web page looking at it *smile*. 
=> So why you do not "smothen" the data there? It's much easier.
=> Anyhow the device is not so accurate, that you can measure 1W, so add a line that shows 0w for any wattage below 5...

BlackAlpha700, 2018-01-02:
My aim is not to measure 1W precise. My aim is to eliminate peaks. Even if I'm replacing wattage below 5W with 0W a peak of 90W would be still reported.
Of course it is possible to smoothen the values in openHAB, but doing so it uses much more ressources of openHAB and the MySQL database than using unused cpu ressorces of the POW.

My openHAB installation controls 270 items and uses about 160 rules. So it is very nice to avoid unimportant rules. ;-)
I'm using the POW to measure if a certain device is switched on and/or running (e.g. washing machine).

lobocobra, 2018-04-15:
Ah got it..... it's a while since, but you might want to look at my POW branch:
- https://github.com/lobocobra/sonoff-brenner
It counts the seconds my oil heating is on and calculates how much oil is left in the tank. The trigger is reaching POWERHIGH.

I had the opposite problem, that the POW went under the POWERHIGH level for one second and then my code believed, that the heating was OFF and then ON again.

I updated the code, which now checks that a POWERHIGH is only off, if POWERHIGH is OFF and POWERLOW is ON. 

In your case you probably want to add a rule that only a POWERHIGH status is called, if POWERHIGH is reached for 30 seconds or more.
=> search for lobocobra in the code and you will find all sections you need to change.