arendst/Tasmota #1479: DS18B20 - stop working after 1 or 2 weeks

ionciubotaru, 2017-12-31:
I'm using a electrodragon with 5 x DS18B20 on GPIO14
All 5 works fine for one week, sometimes 2 weeks, but suddenly none of them are discoreverd by Tasmota.
Performing a software reset doesn't solve the problem, only a hardware reset (turn the power off and on) restore the sensors functionality for another 1 or 2 weeks.

Please write any suggestions how to fix this issue.

Frogmore42, 2017-12-31:
How do you have them wired?
Did you add a pull-up on the data line?
How long is the cable?
Do you have a bus configuration or a star configuration?

Mine have not been running long enough before the power outage to say everything is reliable, with certainty, but it all seems fine so far.

I added a 4R7K resistor from the data line to 3V3 to provide a good pull-up.  I have a 3-4m cable and then a star connection with ~1m cables to 4 sensors.  I have another one that has two sensors in a bus configuration with about 6m of cable.  Both have been working reliably so far.  I did have to add a new connector to a TH16 to get it to work reliably.

ionciubotaru, 2018-01-01:
> How do you have them wired?

1m cable, than all 5 spread like a star with 1m cable each

>Did you add a pull-up on the data line?

Yes, 4K7

>Mine have not been running long enough before the power outage to say everything is reliable, with certainty, but it all seems fine so far.

All of them work fine for 1-2 weeks, so there is no major hardware problem.
I'm sure is something subtile, but I have no idea where to look for.

Frogmore42, 2018-01-01:
I would start with the connection to the  electrodragon.  Wiggle it while starting up to see if things still work.  It might be loose.  Or try using a different gpio pin.  Or try using a different electrodragon/esp8266.  I have a couple of NodeMCU devices that don't work reliably, not sure why yet.  Since I have others that work great, I just don't use the ones that don't work.

eXCepT74, 2018-01-01:
Hello. I have same issue. im using Wemos D1 mini. Uploaded last firmware and tried with DS18b20 and am2301 ( two of the first or two of the second ) but after connecting first one to the bus - on the web appear temp and humidity ( or only temp regard sensor ) , but when connected same type second one - the reading disapearing. i have it on proto board on my tabble. So there is no matter star/bus config. I use 4.7k as pullup. And powering sensors from 3.3v of wemos. Cables have max 25cm. I tried many gpio's and 3 wemos boards so maybe its some wemos problem.

Frogmore42, 2018-01-03:
If you are using multiple DS18B20 devices you have to select the alternative configuration that supports multiple.  It worked fine on my wemos d1 mini, but I am also a few revisions back.

ionciubotaru, 2018-01-03:
I already did it. As I wrote before it works fine for 2 weeks, that means the configuration is OK, the hardware is OK.

arendst, 2018-01-03:
The current version has two versions of DS18x20 drivers to be used with multiple DS18x20 sensors. 

You might want to try the other driver and see how that one performs over time. Selection is done in user_config.h

ionciubotaru, 2018-01-03:
Thank you @arendst 
I used `#USE_DS18x20` till now.
I'll switch to `#define USE_DS18x20_LEGACY` and I hope to send a positive feedback after one month

pascalsaul, 2018-01-03:
I don't have that problem with the same hardware set-up. I've connected five DS18B20's on a Sonoff Dual with the new driver also...

Frogmore42, 2018-01-21:
I have been running 5.10.0b and for 509 hours it was working fine.  But around hour 510 (the first time it has been running this long) all four DS18B20 sensors started reporting 0, which is reported as 00.0 to MQTT and Node-Red didn't like it.  Before rebooting it, I looked at it to make sure there was no loose connection but did not find anything obvious and it was still reading 0.0.  So I just rebooted it an it started working fine.  I'll keep an eye on it to see if it does it again.  I have 4 sensors on about 10m of cable.  This is on the Sonoff TH16 which I added a 3.5mm TRRS connector, so I am pretty sure it is a good one.

Frogmore42, 2018-05-21:
I now have more data.  This no longer seems to be an issue.  I have two devices running for over 800 hours (the last power failure) with no issues.

closed 2018-06-05 by ionciubotaru

vnorbix, 2018-12-04:
I got the same problem as above. I'm using a Wemos mini D1, 2 pieces of DS18B20 are connected to it on a bus (~1m to the first, ~10cm to the second), I have a resistor as well, I checked connections.  I set the tele period to 1 minute, because I needed frequent data. Everything works perfectly until a few days, then sensor 1 (closer to the board) doesn't send data anymore. This happened twice already, last time it was 3 days until it disappeared. After I turn on debug mode I see this:
```
07:55:29 DSB: Sensor CRC error
07:55:29 SNS: DS18B20-1 missed 5
```
Sensor 2 is still sending data then.

I tried restart sonoff from the web interface and cut power line to the sensors temporarily,
but only powering down the board helped and then temperature data showed up again from sensor 1.

Do you have any suggestions what can be the problem here or what should I look for? I'm using version 6.3.

Frogmore42, 2018-12-04:
You didn't include any information about exactly what you are using for software, ie there are at least two versions of the driver for the DS18B20.  But, based on what you are seeing it is possible one of your sensors is bad/marginal.

The error I was seeing was resolved by restarting via the web UI, which indicated it might be software related.  When the error went away with newer software, it was pretty clear it was software.

But, I have had DS18B20 devices fail.  All of the ones that have failed have been due to corrosion on/of the connections, despite some of them claiming to be waterproof, they are only resistant.

vnorbix, 2018-12-04:
I added the `` -DUSE_SENSORS`` switch in platformio.ini.

If it's hardware related, how can be that it works without problem until some time, then it just stops working for good, until I do a powerdown on the board. I don't fiddle with cables and connections when doing a powerdown.

Actually I got the same problem with a DHT22 earlier which I used with a sonoff relay. I thought there the sensor might have been defective or just unreliable and DS18B20 will be stable. The difference was there I could got back the data once I cut the power from the sensor.

Frogmore42, 2018-12-04:
That is the exact signature of some types of defects.  It works fine, until it doesn't.  The software is indicating there are communication errors.  Since a simple reset of the software doesn't resolve the problem, it is likely an issue with the hardware.  If it was mine I would replace the sensor and see if the problem goes away.

vnorbix, 2018-12-05:
Thank you, I'll try to replace it.