arendst/Tasmota #1481: Sonoff S20 not connecting to WIfi

SGF-Lon, 2017-12-31:
I have been trying (unsuccessfully) to get my recently flashed Sonoff S20 to connect to either of my Wifi AP.  When switching on LOG_LEVEL_DEBUG messaging for the serial console, I can see it waiting & looping for the connection to be established ("Attempting connection.....") with no success.  The 'error message' is "Connect failed with AP timeout". And then it tries the alternate SSID with the same failed result.  There are no unusual, error or rejection log entries in the router log.  Any suggestions on how to debug further?

Background:
   Using Sonoff-Tasmota 5.10.0 - downloaded on 31/12/17
   Built & uploaded successfully using PlatformIO (downloaded on 31/12/17)
   Successfully used SONOTA OTA to upload Tasmota 5.10.0 using the precompiled binaries onto a
         Sonoff Touch & 2xSonoff POW.  They all connect to both networks with no problem.
   Changed WIFI_CHECK_SEC & WIFI_RETRY_SEC to see if longer intervals made any difference
          (they didn't).
   Changed WIFI_CONFIG_TOOL to WIFI_WAIT & WIFI_MANAGER to see if they changed the behaviour.
         (they did as expected - but still no connection).
   Tried using WIFI_MANAGER mode and connecting via AP & then using webservices to change the 
         SSIDs & Passwords & then reboot.  Still won't connect.
   Tried using invalid passwords and the Error messaging is different, as expected.
        Connection rejected due to invalid password.
   The router is a standard BTHomeHub5 in the UK, configured to support 802.11 b/g/n
   The logging message indicates that Tasmota is attempting to connect in MODE 11N.
   I can see the logging message PATCH 2861 just before the attempted connection.
   When using OTA to flash these S20 devices, they refuse to connect back to the network in order
         to receive the payload.
   This device work(ed) perfectly well with their stock firmware & eWeLink.  No trouble connecting to
       either network for discovery or usage for several weeks.

Any help, insight or suggestions would be welcome, as I can't

davidelang, 2017-12-31:
are you sure you have enough signal strength and are using the correct ssid and 
passord?

SGF-Lon, 2017-12-31:
Thanks for the suggestion.  Yes, the SSID & Password are correct (verified several times and hard-coded into the binary which is used by the other Sonoffs (which do connect).  
Signal strength has been different at different times.  Tests have included right next to the router, near the router, in another room, downstairs - and all to no success.  Despite other Sonoff devices working in all of these locations.  I'm starting to wonder if there is something wrong with the actual physical device.

Anyone know of a way to do more detailed logging (status bits, etc) to see what's happening in the chip/firmware?

webtop, 2018-01-01:
I don’t suppose you have either a Raspberry Pi or Linux box? If so, set one of those up as a DHCP server and try to connect to that. You will be able to do much more verbose logging that way. And if it fails to receive a connection request, you’ll know the issue is the device or firmware. If that’s the case, load the same exact firmware onto another device and see if that will connect to the new DHCP server. If it succeeds then you know it’s not the firmware. That only leaves the device, so you’ll know it’s going to be hardware troubleshooting.

avrhack, 2018-01-01:
Check https://github.com/arendst/Sonoff-Tasmota/issues/1471

There was a PlatformIO update that bricked the firmware - simple fix in that chain by changing platformio.ini line to be

platform = espressif8266@1.5.0

It may not be your problem, but worth ruling it out by making that change, doing a clean build and reflashing the device. My s20 was completely unstable and would not connect, just like yours, until I did that.....

webtop, 2018-01-02:
I had the same bricking issue, but didn't change my .ini settings, and have no problems...but I'm not flashing to a Sonoff device, just a raw 12 module

SGF-Lon, 2018-01-06:
Thanks everyone who replied....  Webtop - you were right!  It seems that the problem was DHCP not responding properly to the connection request for an IP address.  So, the S20 never connected to the network, etc.  I solved this problem by installing DNSMASQ on my Raspberry PI (good idea!) and enabling full logging.  And there it was - very easy to spot  So, I have switched off DHCP on my BTHomeHub5 permanently and have moved the entire home LAN over to DNSMASQ. And everything is working better & faster. 
So, there wasn't a problem with the Tasmota firmware nor the ESP8266 chip.  Just a reluctant network component - which has now been retired.
This can be closed.

closed 2018-01-06 by SGF-Lon

jmcbri, 2018-05-20:
Per another forum:

SonoffBuyer
said 3 months ago

I registered an account just so I can help others on getting a Sonoff paired.

After trying for hours and using AP mode, pairing from fresh power up, trying to pair to WiFi without special characters in, pairing to a mobile phone hot spot, etc. etc.  The ONLY thing that worked STRAIGHT away, was to put the phone into airplane mode, then only activating the wifi to connect with home router, then running the app to pair, and it works!!

1 person likes this

I suspect that having bluetooth enabled may make it think there's an alternative connection method.

eihnat, 2018-07-18:
nothing worked for me to pair. airplane mode hint worked. so +1 on that.