arendst/Tasmota #1483: PZEM004T no data output

ihuex, 2018-01-01:
pzem004t use tasmota 5.10.b/g , diy esp8266  tasmota ,data have output a time ,then no data output all is 0V 0A 0W 0W 0 W , i switch pzem004t rx tx to espeasy ,is ok, so why ?

ionciubotaru, 2018-01-01:
Please provide more details - pzem004t has a TX/RX output, you can read data using a FTDI via USB port.
I suppose you want to upload the pzem004t output to a MQTT server using a ESP8266.
By default Tasmota doesn't do this, you must disable Tasmota's serial, than modify the code.

ihuex, 2018-01-01:
,yeah.i use esp 8266 gpio0 gpio2 to pzem004t rx tx,  It can be used in the beginning.but now no data output all is zero ,,,so i use other esp8266 falsh espeasy,by use same gpios, can read , yeah，i use windows app，also data output...

gajotnt, 2018-01-01:
Are you giving the pzem 5v? Show how you connect it.

ihuex, 2018-01-02:
i add 220-5v module  and use 5v-3.3 to esp8266

orsgah, 2018-05-07:
Same problem, on sonoff rm not working, but on nodmcu work fine. I replace resistor to work on 3.3v. Then i try to use power from sonoff PCU and TX RX to nodemcu, an all work fine. Then conect back RX TX to sonoff and nothing.

closed 2018-07-06 by stale[bot]