arendst/Tasmota #1484: SONOFF POW rebootin all 17-23 seconds

lobocobra, 2018-01-01:
Hi all

I need your support please.

**Setup**
- I flashed a SONOFF POW with the latest software  5.7.1
- I connected my heating on it an it uses 1.4 ampere and up to 240kwh

**Issue statement**
All 17-23 seconds sonoff POW is rebooting and shows an exception error:
15:35:46 MQT: tele/ug_heizung_brenner/INFO1 = {"Module":"Sonoff Pow", "Version":"5.7.1", "FallbackTopic":"ug_heizung_brenner_B447CA", "GroupTopic":"sonoff-basic"}
15:35:46 MQT: tele/ug_heizung_brenner/INFO2 = {"WebServerMode":"Admin", "Hostname":"ug_heizung_brenner-1994", "IPAddress":"192.168.20.198"}
15:35:46 MQT: tele/ug_heizung_brenner/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4020290c epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
15:35:47 MQT: stat/ug_heizung_brenner/RESULT = {"POWER":"ON"}

The only good thing is that the heating is not shutoff during this fast CLICK reboot.
=> can anyone help me please to find a solution

arendst, 2018-01-01:
Try the real latest release 5.10.0

lobocobra, 2018-01-01:
Thanks for the fast replay, I am on it. Currently fighting with the error....
=> fatal error: TasmotaSerial.h: No such file or directory  #include <TasmotaSerial.h>

I must have something overlooked in the wiki, when I found it and testet it, I will report it here.

arendst, 2018-01-01:
It's not in the wiki yet. Copy it from the lib folder.

lobocobra, 2018-01-01:
Feedback: Issue could be solved by upgrading to version 10g.
=> Thus the error was fixed during the last updates.

closed 2018-01-01 by lobocobra

lobocobra, 2018-01-03:
Thanks for the hint with the Wiki. It was not written there, but I did update the Wiki.
=> So it should be clearer now for others as well.