arendst/Tasmota #1487: compiling issue

mrmchaffey, 2018-01-02:
I keep getting this message and cant find a answer on the web.

 Arduino: 1.8.1 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DOUT, 115200, 512K (64K SPIFFS), ck, Disabled, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Program Files (x86)\Arduino\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dout,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -vid-pid=0X2341_0X0001 -ide-version=10801 -build-path C:\Users\office\AppData\Local\Temp\arduino_build_15952 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\office\Desktop\Sonoff-Tasmota-5.10.0\Sonoff-Tasmota-5.10.0\sonoff\sonoff.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Program Files (x86)\Arduino\portable\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Program Files (x86)\Arduino\portable\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Program Files (x86)\Arduino\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dout,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -vid-pid=0X2341_0X0001 -ide-version=10801 -build-path C:\Users\office\AppData\Local\Temp\arduino_build_15952 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\office\Desktop\Sonoff-Tasmota-5.10.0\Sonoff-Tasmota-5.10.0\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\office\AppData\Local\Temp\arduino_build_15952/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\office\AppData\Local\Temp\arduino_build_15952\sketch\sonoff.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\office\AppData\Local\Temp\arduino_build_15952/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Program Files (x86)\Arduino\portable\sketchbook\libraries\PubSubClient\src" "C:\Users\office\AppData\Local\Temp\arduino_build_15952\sketch\sonoff.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\office\AppData\Local\Temp\arduino_build_15952/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Program Files (x86)\Arduino\portable\sketchbook\libraries\PubSubClient\src" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" "C:\Users\office\AppData\Local\Temp\arduino_build_15952\sketch\sonoff.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\office\AppData\Local\Temp\arduino_build_15952/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Program Files (x86)\Arduino\portable\sketchbook\libraries\PubSubClient\src" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "C:\Users\office\AppData\Local\Temp\arduino_build_15952\sketch\sonoff.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\office\AppData\Local\Temp\arduino_build_15952/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Program Files (x86 [... cut ...]
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\Wire 
Using library IRremoteESP8266 at version 2.3.1 in folder: C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\IRremoteESP8266 
Using library NeoPixelBus_by_Makuna at version 2.2.9 in folder: C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\NeoPixelBus_by_Makuna 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI 
exit status 1
Error compiling for board Generic ESP8266 Module.

webtop, 2018-01-02:
Looks like your project is too big for the board type you used. What version of board do you have? If it's a 12/12e, set your board type to nodemcuv2 in the platformio.ini file.

mrmchaffey, 2018-01-02:
Sonoff TH_v1.1  2017-5-5

it never makes it past the compiling step, sonoff is not even attached to the pc

mrmchaffey, 2018-01-02:
i have the somoff basic

arendst, 2018-01-02:
The problem is in your second line.

The answer is in the readme.md and the wiki.

Look for some seamingly cyptic name like "linker script"

mrmchaffey, 2018-01-02:
Solved.  The memory chip on the board is a pn25f08b it's only found on the newer boards ,all I did was switch the board too generic esp8285 and it compiled and flash ....my setup is a arduino uno with the reset pin hooked to ground and a sonoff basic esp8266 chip and a pn25f08b 1mb chip .     Hope this helps you guys.    And thanks for the quick response's

mrmchaffey, 2018-01-02:
Also I had a problem downloading the json library in ide it kept giving me a red message at the bottom of the screen something about a download error I assumed it was my internet or the GitHub server turns out all you have to do is close Arduino IDE and right click on it open it as administrator and it works fine just putting that out there cuz I couldn't find any info on that on the web just happened to stumble across that figuring it out last night